raspberrypi/pico-vscode #125: Failed to download and install tools.

AndrewCapon, 2024-11-25:
Hi Guys,

I thought I would give the extension a go.

I already have pico-idk and the tools installed but these are ignored and it attempts to download them again.

"Failed to download and install tools." is displayed as an error.

a `.pico-sdk` folder is created in my home containing:

```
andrewcapon@iMac .pico-sdk % ls -ltr
total 0
drwxr-xr-x  3 andrewcapon  staff  96 25 Nov 07:03 cmake
drwxr-xr-x  3 andrewcapon  staff  96 25 Nov 07:03 sdk
drwxr-xr-x  3 andrewcapon  staff  96 25 Nov 07:03 toolchain
drwxr-xr-x  3 andrewcapon  staff  96 25 Nov 07:04 tools

andrewcapon@iMac .pico-sdk % ls cmake
pico-vscode.cmake

andrewcapon@iMac .pico-sdk % ls sdk
2.1.0

andrewcapon@iMac .pico-sdk % ls toolchain 
13_3_Rel1

andrewcapon@iMac .pico-sdk % ls tools
2.1.0

```

fabianoriccardi, 2024-11-25:
Same here. Using on VScode 1.95.3; win10 22H2 and extension v0.17.1.

AndrewCapon, 2024-11-25:
Ah, I'm not alone :)

mahesh-slx, 2024-11-25:
It was working for me until last Friday with SDK version 2.0.0. However, the Pico extension auto-updated today, and the SDK is now at version 2.1.0, and I’m encountering the same issue.

will-v-pi, 2024-11-25:
Should be fixed now - you can download the VSIX from https://github.com/raspberrypi/pico-vscode/actions/runs/12009019275 to test it now, or just wait for the new release later today

will-v-pi, 2024-11-25:
Release has been published, so if you update your extensions this issue should be fixed

AndrewCapon, 2024-11-25:
Thanks will.

Is there anyway of telling it to use copies of the toolchain, tools and sdk that I already have installed?

will-v-pi, 2024-11-25:
No, this extension downloads its own copies of everything when creating the project. Once the project has been created you can manually edit the CMakeLists.txt file to use your own tools if you want though.

AndrewCapon, 2024-11-25:
Thanks for the info, maybe a feature request for the future would be to have some options to set the paths for these?

will-v-pi, 2024-11-25:
Feel free to create a separate issue for that feature request, detailing which paths you want options to set manually

fabianoriccardi, 2024-11-25:
Great, the patch works.

will-v-pi, 2024-11-26:
Thanks - I will close this issue as fixed with 0.17.2

closed 2024-11-26 by will-v-pi

nonchip, 2025-07-01:
this is still happening for me on 0.17.6, on VSCodium 1.101.24242.glibc2.29-x86_64.AppImage

will-v-pi, 2025-07-01:
This particular issue was related to the new SDK release, and was fixed. If you’re having separate issues, please file a new bug report, and fill in the template with as much information as you can provide