raspberrypi/pico-vscode #126: OSX Intel OpenOCD?

AndrewCapon, 2024-11-26:
Hi Guys,

I notice in another thread that there is no Intel OpenOCD downloaded with this plugin.

I just wondered why this is?

Many thanks

Andy

will-v-pi, 2024-11-26:
This is simply because we haven't been able to get the Intel OpenOCD to compile on Arm64 MacOS, as we're currently only using Arm64 GitHub runners (see the [pico-sdk-tools](https://github.com/raspberrypi/pico-sdk-tools) repository). For picotool and pioasm we are able to cross-compile universal binaries, but for OpenOCD we were unable to get the cross-compile working correctly.

I'll look into compiling an Intel OpenOCD on an Intel MacOS runner, which would fix this problem

AndrewCapon, 2024-11-26:
Hi Will,

Thanks for the info.

I did have to edit the launch.json files to get them going here but I'm guessing it is something to do with the fact that it isn't using the supplied OpenOCD. So basically just pointing it at the correct OpenOCD binary and using `-s`.

I'll give it all a test again when the OpenOCD is available.

Really useful Extension by the way, now I don't have to copy and paste stuff from existing projects to new projects to get them going:)

Nice job.

Cheers

Andy