raspberrypi/pico-vscode #127: had to run git submodule update --init in 2.1.0 SDK directory

jdbruner, 2024-11-27:
I switched the SDK for an existing project from 2.0.0 to 2.1.0.  The vscode extension downloaded the new SDK (and updated toolchain). However, when I attempted to reconfigure and rebuild the project, there were multiple complaints about missing files. It was necessary for me to cd to ```~/.pico-sdk/sdk/2.1.0``` and run ```git submodule update --init``` manually to resolve them. After that, everything configured and built successfully.

will-v-pi, 2024-11-27:
The extension should check out the submodules automatically for 2.1.0, just like it does for 2.0.0 - did you see any errors saying the SDK download failed when you switched the SDK?

To double check if this is intermittent or not, could you also try deleting the `~/.pico-sdk/sdk/2.1.0` directory and then re-opening you project - the extension should then clone the SDK again and update the submodules, or throw an error if the update fails

jdbruner, 2024-11-27:
I should have noted originally that this particular issue is on Windows 11 (x64).

I removed the 2.1.0 directory and reloaded the project. It complained that python3 is not installed or in the PATH. However, it is present and is in my ```%PATH%``` - I can run it from the command line as either ```python3.exe``` or ```python.exe```  Where does the extension expect it to be?

Incidentally, [Getting Started with Raspberry Pi](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) section 3.1.1 says that no dependencies are needed for Raspberry Pi OS or Windows. (Section 3.1.2 says that Python 3.9 or later is required for Linux.) If this has changed, then you may want to update that document also.

will-v-pi, 2024-12-02:
Is there a python executable present under `~/.pico-sdk/python/3.12.1`? It's correct that no dependencies are required on Windows, because the extension should install it's own version of python to that location and use that if there is no other python present, so it sounds like installation of that on your system might have gone wrong

kevinjuszczyk, 2024-12-02:
I ran into this issue as well. FWIW, I don't have a ~/.pico-sdk/python/ directory at all, but I do have a working Python 3.12.7 installed on the system.

will-v-pi, 2024-12-02:
> I ran into this issue as well. FWIW, I don't have a ~/.pico-sdk/python/ directory at all, but I do have a working Python 3.12.7 installed on the system.

Just to confirm, was your issue with the submodules not being present, or the error that python3 was not installed?

If you already have python then the extension should use that, rather than installing it's own, hence there being no `~/.pico-sdk/python/` directory

kevinjuszczyk, 2024-12-02:
The issue was with the modules not being present. 
I did have the SDK installed separately, and the PICO_SDK_PATH variable set to that. I was trying to build the debugprobe project and it was unable to find the tinyusb library. I was confused by this, since the tinyusb files were present in the above path, but I eventually realized that ~\.pico-sdk\sdk\2.1.0\lib\tinyusb was where I should have been looking; this path had no files in it. After running 'git submodule update --init' under ~\.pico-sdk\sdk\2.1.0 I was able to successfully build.

jdbruner, 2024-12-02:
On my machine there is an installation of Python in ```.pico-sdk\python\3.12.1```  However, it looks like the extension is finding the one that is installed on my machine. When I look at its settings, I see that ```raspberry-pi-pico.python3Path``` is ```...\AppData\Local\Microsoft\WindowsApps\python3.12.exe```  I note that this is not a regular file but instead is a reparse point. I wonder if that could be significant...?

jdbruner, 2024-12-04:
I manually changed ```raspberry-pi-pico.python3Path``` to point to the Python installation in ```.pico-sdk```, deleted the 2.1.0 SDK, and opened my project folder. It successfully (re-)installed the 2.1.0 SDK without errors, and it looks like it handled the submodules (so it must have successfully run python).

Therefore, as I noted previously, I wonder if the issue is that pathname it is using for python3.12 is a reparse point.

will-v-pi, 2024-12-04:
It's possible that's the issue - also note that in #94 (released in 0.17.0) we changed the python handling to be machine-scoped, so instead of having `raspberry-pi-pico.python3Path` in every project's `settings.json` file, it's now stored in the VS Code User Settings.

If you were changing the setting in the project's `settings.json` file, you should be able to delete that line entirely, and the extension will then use the VS Code Python extension to locate the Python installation, which should work better

jdbruner, 2024-12-04:
I tried that, and the python locator set ```python3Path``` to the one in ```...\AppData\Local\Microsoft\WindowsApps\python3.12.exe```. That's the one that causes the pico extension to fail. My workaround was to manually edit the path to point to the installation in ```.pico-sdk```

I think this is an issue that is specific to the pico extension, because otherwise python works fine.

will-v-pi, 2024-12-17:
@jdbruner Is your Python issue possibly related to https://github.com/raspberrypi/pico-vscode/issues/136 (if you have your language set to something other than English (US))?

will-v-pi, 2024-12-17:
In which case, I'm afraid I'm unable to reproduce your issue. For me, when using the Microsoft Store python 3.12 (which I believe is what you're using based on the paths provided), and with the `python3path` set to the reparse point at
```
c:\Users\<user>\AppData\Local\Microsoft\WindowsApps\python3.12.exe
```
the extension is working without any issues for me.

will-v-pi, 2025-01-06:
Closing this as unable to reproduce the python issue, and the original git submodule issue has been fixed

commit 2e39516f (2025-01-23): Don't use Windows Store python

Do not pick the Windows Store python, and clear it from the settings if it was already chosen, as it seems to cause issues on some systems

Fixes #127 and #156
closed 2025-01-06 by will-v-pi