raspberrypi/pico-vscode #129: Output for CMake actions?

AnalogWeapon, 2024-12-04:
I'm presenting this as a "problem", but it might be that I don't understand where to look: When the "Clean CMake" and "Configure CMake" tasks are run, there is no output besides the little notification in the lower-right. I want to see the output from the CMake process, so I can can figure out what's going on and make changes if needed. I don't know what files it is referencing to execute CMake, so that's the first thing I'm trying to figure out.

If something goes wrong, the notification bubble will advise to "check the logs". Where are those logs? I have searched through all the possible output sources in vscode's output tab. I don't see any terminals created either. I don't see anything popping up in the developer tools of vscode either.

will-v-pi, 2024-12-04:
To get more output from CMake, I’d recommend using the [CMake Tools Integration] (https://github.com/raspberrypi/pico-vscode?tab=readme-ov-file#cmake-tools-extension-integration) - this will give you all the logs from the CMake Tools extension, and that will provide a lot more configurability for the CMake build

AnalogWeapon, 2024-12-05:
I have enabled those settings, reloaded the window, and restarted vscode. No output shows when I click "Clean CMake" or "Configure CMake" from the Raspberry Pi Pico extension pane on the right. When you say that it gives the logs, do you mean what is generated in the `build` folder? I'm looking for output from the running processes.

will-v-pi, 2024-12-05:
With the CMake Tools integration enabled you'll need to use that to run the configure/clean - simplest way is with Ctrl+Shift+P "CMake: Configure". The output appears in the Output window (Ctrl+Shift+U).

You should get an error mentioning this when you click the "Configure CMake" button from this extension (when you have the CMake Tools Integration enabled), saying that you must use the CMake Tools extension.

AnalogWeapon, 2024-12-05:
That makes sense. The output does appear when I use CMake Tools.

So I guess the only problem I'm having is that that error isn't popping up for me when I click "Configure CMake". It just continues to show the notification of success or failure. But that's a pretty trivial error, knowing that I should just use the CMake Tools extension for those functions instead.

will-v-pi, 2024-12-05:
Just to confirm, have you definitely changed the `raspberry-pi-pico.useCmakeTools` setting in `settings.json` in your project directory from `false` to `true` and reloaded the VS Code window?

AnalogWeapon, 2024-12-05:
I *swear* that I had done that while testing yesterday, but when I went to confirm again now, it is behaving as you expected: I am getting the warnings that I should use CMake Tools instead. I'm sorry for the confusion. Appreciate the help.

Stokestack, 2024-12-07:
I just encountered the same problem: "Cmake failed to configure your build." but no way to find out why. If I run CMake from the command line, it reports success.

Which "Cmake Tools Integration" extension do you mean? I have the Microsoft "CMake Tools" installed and enabled.

Do not use "CMake Integration" by Christoph Seitz; it blanks out your environment variables for some reason, so the paths in your build configuration files will break: https://github.com/go2sh/cmake-integration-vscode

will-v-pi, 2024-12-07:
> Which "Cmake Tools Integration" extension do you mean? I have the Microsoft "CMake Tools" installed and enabled, but I'm not seeing any options to get useful information.

As detailed in the readme, it’s the “CMake Tools Extension” integration, which integrates Microsoft’s CMake Tools Extension with this extension.

> For some reason, running the Pico extension's "Clean CMake" option blew it away, and it doesn't get rebuilt.

The “Clean CMake command is supposed to blow it away (that’s the whole point of the clean), but it should reconfigure it after. It sounds like your CMake configuration isn’t correct (possibly an error in your CMakeLists.txt file), so you should enable the CMake Tools Extension integration to get more output to see what’s going wrong with the CMake configure.

Stokestack, 2024-12-07:
Thanks. I restored my project from a backup and tried another clean. This time it didn't blow away the Ninja file, so I don't know what happened last time. I'm using the CMake extension to build now, and the Pi one to debug. It's working fine at the moment.

will-v-pi, 2024-12-17:
Leaving this open to maybe improve the error message so it's clearer which logs to check, and mention the CMake Tools Extension integration

commit d6a4a1a4 (2025-01-15): Improve error message when CMake Configure/Clean fails

Fixes #129
closed 2025-01-15 by will-v-pi