raspberrypi/pico-vscode #130: pico-extras

alidaf, 2024-12-05:
I can't find any information on how to set up pico-extras using this extension. Can pico-extras be included in the extension itself? Alternatively, a brief guide on how to add it here would be very useful. I've tried multiple sources but they are either out of date or require additional steps not covered.

will-v-pi, 2024-12-05:
You can just include it in the way described in the README - clone the repository, set `PICO_EXTRAS_PATH` in the CMakeLists.txt to that path, then copy the `external/pico_extras_import.cmake` to your project directory and include it after `pico_sdk_import.cmake`.

You can also do this rather than copying the file into your project - add these after the `include(pico_sdk_import.cmake)` line in CMakeLists.txt:
```cmake
set(PICO_EXTRAS_PATH /path/to/pico-extras)
include(${PICO_EXTRAS_PATH}/external/pico_extras_import.cmake)
```

alidaf, 2024-12-05:
None of those work for me, which is why I opened the issue.