raspberrypi/pico-vscode #133: Windows arm64 Version error message "There is no data provider registered that can provide view data."

3ueb, 2024-12-06:
Hello,

I tried to install the plugin. VS Code shows me the following error: 

There is no data provider registered that can provide view data.

A popup from MicroPico gave me the following error:

No COM device found. Please check your connection or ports and try again. Alternatively you can set the manualComDevice setting to the path of your COM device in the settings but make sure to deactivate autoConnect. For Linux users: check you sufficient permission to access the device file of the Pico.

In the Developer Tools of VS Code I found this error message:

Cannot activate the 'Raspberry Pi Pico' extension because its dependency 'Serial Monitor' failed to activate

and 

notificationsAlerts.ts:42 No COM device found. Please check your connection or ports and try again. Alternatively you can set the manualComDevice setting to the path of your COM device in the settings but make sure to deactivate autoConnect. For Linux users: check you sufficient permission to access the device file of the Pico.

I tested it on an x64 PC, and it works so far. Can anyone help or suggest what I can do?

VS Code Version:
Version: 1.95.3 (user setup)
Commit: f1a4fb101478ce6ec82fe9627c43efbf9e98c813
Date: 2024-11-13T14:50:04.152Z
Electron: 32.2.1
ElectronBuildId: 10427718
Chromium: 128.0.6613.186
Node.js: 20.18.0
V8: 12.8.374.38-electron.0
OS: Windows_NT arm64 10.0.26100

will-v-pi, 2024-12-06:
Yes, it’s in the [readme](https://github.com/raspberrypi/pico-vscode?tab=readme-ov-file#requirements-by-os) - only Windows x64 is supported, not Windows on Arm

closed 2024-12-06 by will-v-pi

3ueb, 2024-12-07:
The good thing is, it works if I install the VS Code in the x64 Version. The best is, to install it in the portable mode https://code.visualstudio.com/docs/editor/portable