raspberrypi/pico-vscode #134: Visual Studio Code Dev Container Support

bewster, 2024-12-12:
A very promising extension.

It would be nice to eventually have a version of this extension that supported\worked inside a container.
With option to autogenerate the `devcontainer.json` by the extension (from new or example project wizard).

Official docker containers containing the sdk, toolchain, could be useful.
Then to not have to have all these files saved on the host machine (under `%USERPROILE%\.pico-sdk`).

Reading USBIPD shows some promise on connecting to hardware; but native USB support may be some way off.

mwinters-stuff, 2024-12-17:
you could look at my repo (doesnt have this plugin) could be used as an example

https://github.com/mwinters-stuff/vscode-devcontainer-raspberrypi-pico

Probably quite easy to add it in, i have only just been trying this extension for the last day.

bewster, 2024-12-17:
> you could look at my repo (doesnt have this plugin) could be used as an example
> 
> https://github.com/mwinters-stuff/vscode-devcontainer-raspberrypi-pico
> 
> Probably quite easy to add it in, i have only just been trying this extension for the last day.

Thanks, there a few examples out there (I thought about including them in my original post).
https://github.com/dodancs/pico-sdk-dev-container
https://github.com/cyberkaida/raspberry-pi-pico-devcontainer
(There may be more)

will-v-pi, 2024-12-17:
This is not a priority, but we'd be open to PRs to add this. Note that none of those devcontainers linked install picotool correctly for the SDK to find it (see the [picotool README](https://github.com/raspberrypi/picotool?tab=readme-ov-file#usage-by-the-raspberry-pi-pico-sdk))