raspberrypi/pico-vscode #135: Clicking on "Clean CMake" removes "build/compile_commands.json" in C++ project

qubit999, 2024-12-13:
Hello, 

is this a feature or a bug?

I have the same issue on Mac and on Windows.

will-v-pi, 2024-12-13:
The Clean CMake button simply deletes the entire build directory, then re-runs CMake to recreate it. It should recreate the `compile_commands.json` file after it has run, along with the `build.ninja` file and all the other project configuration.

Is your issue that it's not recreating the `compile_commands.json` file after reconfiguring your build diresctory?

qubit999, 2024-12-13:
Exactly.

qubit999, 2024-12-13:
I've created with the extension. I will send you later the day. I'm not at my computer rn.

iodar, 2024-12-13:
I might be able to give some info here. Seen the same behaviour, but realised it only appears once you have run `cmake .` from project root. If you run `Clean CMake` from `pico` extension afterwards, it will remove the `build` folder.

## Init project
Init project from `pico` extension
[image]
[image]
[image]

### Attachments
- [CMakeLists.txt](https://github.com/user-attachments/files/18132943/CMakeLists.txt)
- [pico-after-init.tar.gz](https://github.com/user-attachments/files/18132946/pico-after-init.tar.gz)

## Compile using CMake
Compile project using CMake Compile option from `pico` extension
[image]
[image]

### Attachments
- [pico-after-compile.tar.gz](https://github.com/user-attachments/files/18132947/pico-after-compile.tar.gz)

## Run `cmake .`
Run `cmake .` command in projct root
```
-- Defaulting build type to 'Release' since not specified.
-- The C compiler identification is GNU 13.3.1
-- The CXX compiler identification is GNU 13.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/<username>/.pico-sdk/toolchain/13_3_Rel1/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/<username>/.pico-sdk/toolchain/13_3_Rel1/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/<username>/.pico-sdk/toolchain/13_3_Rel1/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python3: /usr/bin/python3 (found version "3.11.2") found components: Interpreter 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/<username>/workspace/pico-sample-project
``` 

[image]

### Attachments
- [pico-after-cmake-cli-command.tar.gz](https://github.com/user-attachments/files/18132952/pico-after-cmake-cli-command.tar.gz)

## Clean project from `pico` ext

Clean project from `pico` extension
[image]

[image]

Error message pops up
[image]

Folder `build` is gone

### Attachments

- [pico-after-clean-after-cli-cmake-run.tar.gz](https://github.com/user-attachments/files/18132954/pico-after-clean-after-cli-cmake-run.tar.gz)

qubit999, 2024-12-14:
This is my CMakeLists.txt:

```
# == DO NOT EDIT THE FOLLOWING LINES for the Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.1.0)
set(toolchainVersion 13_3_Rel1)
set(picotoolVersion 2.1.0)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================
# Generated Cmake Pico project file

cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)

# == DO NOT EDIT THE FOLLOWING LINES for the Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.1.0)
set(toolchainVersion 13_3_Rel1)
set(picotoolVersion 2.1.0)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================
set(PICO_BOARD pico2_w CACHE STRING "Board type")

# Pull in Raspberry Pi Pico SDK (must be before project)
include(pico_sdk_import.cmake)

project(first_project C CXX ASM)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

# Add executable. Default name is the project name, version 0.1

add_subdirectory(Pico_code/c/lib/Config)
add_subdirectory(Pico_code/c/lib/Fonts)
add_subdirectory(Pico_code/c/lib/GUI)
add_subdirectory(Pico_code/c/lib/Icm20948)
add_subdirectory(Pico_code/c/lib/Infrared)
add_subdirectory(Pico_code/c/lib/LCD)
add_subdirectory(Pico_code/c/lib/OLED)
add_subdirectory(Pico_code/c/examples)

include_directories(./Pico_code/c/lib/Config)
include_directories(./Pico_code/c/lib/Fonts)
include_directories(./Pico_code/c/lib/GUI)
include_directories(./Pico_code/c/lib/Icm20948)
include_directories(./Pico_code/c/lib/Infrared)
include_directories(./Pico_code/c/lib/LCD)
include_directories(./Pico_code/c/lib/OLED)
include_directories(./Pico_code/c/examples)

add_executable(first_project main.cpp)

pico_set_program_name(first_project "first_project")
pico_set_program_version(first_project "0.1")

# Modify the below lines to enable/disable output over UART/USB
pico_enable_stdio_uart(first_project 1)
pico_enable_stdio_usb(first_project 1)

include_directories(${SDK_PATH}/usr/include)
link_directories(${SDK_PATH}/usr/lib)

# Add the standard include files to the build
target_include_directories(first_project PRIVATE
    ${CMAKE_CURRENT_LIST_DIR}
    ${PICO_SDK_PATH}
)

# Add any user requested libraries
target_link_libraries(first_project 
    pico_stdlib
    hardware_spi 
    hardware_i2c 
    hardware_dma 
    hardware_pio 
    hardware_interp 
    hardware_timer 
    hardware_watchdog 
    hardware_clocks 
    pico_cyw43_arch_none
    Config
    Fonts
    GUI
    Icm20948
    Infrared
    LCD
    OLED
    examples
)

pico_add_extra_outputs(first_project)
```

will-v-pi, 2024-12-17:
> I just noticed that it appended another non-edit block. Guys, can you fix this? I need a working VS code extension without fearing to break the build process by reconfiguring or cleaning CMake. I use this feature everytime I update my code.

It only does this is you import a project that doesn't need to be imported (ie an existing VS Code project, that already has the DO NOT EDIT block). We should probably add an error to stop people doing this though, as it has come up a few times. You should just delete the top DO NOT EDIT block.

Regarding why running CMake from the command line works but the Configure/Clean CMake commands don't work, my guess would be the relative paths in your CMakeLists.txt file. Instead of `./Pico_code/c/lib/Config` you should use `${CMAKE_CURRENT_LIST_DIR}/Pico_code/c/lib/Config` so that the paths are relative to the CMakeLists.txt file.

Also, I'm not sure what these lines are for, but they probably won't work as `SDK_PATH` is not set in your CMakeLists.txt file

> include_directories(${SDK_PATH}/usr/include)
> link_directories(${SDK_PATH}/usr/lib)

Finally, when you say `build/compile_commands.json` is missing, is `build/build.ninja` also missing?

@iodar Don't just run `cmake .` in your project root, as then you get a build in the source tree, which is not supported by this extension. Instead either `mkdir build && cd build && cmake ..` or `cmake -S . -B build` to create the `build` directory for the build. Or just use the commands in the extension (after deleting all the files that you created in your source tree running `cmake .`, which are probably causing the issues you're seeing)

qubit999, 2024-12-18:
This is my updated CMakeLists.txt:

`cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)

# == DO NOT EDIT THE FOLLOWING LINES for the Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.1.0)
set(toolchainVersion 13_3_Rel1)
set(picotoolVersion 2.1.0)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================
set(PICO_BOARD pico2_w CACHE STRING "Board type")

# Pull in Raspberry Pi Pico SDK (must be before project)
include(pico_sdk_import.cmake)

project(first C CXX ASM)

set(PICO_CXX_ENABLE_EXCEPTIONS 1)

set(PICO_CXX_ENABLE_RTTI 1)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

# Add executable. Default name is the project name, version 0.1

add_subdirectory(Pico_code/c/lib/Config)
add_subdirectory(Pico_code/c/lib/Infrared)
add_subdirectory(Pico_code/c/lib/OLED)
add_subdirectory(Pico_code/c/lib/Icm20948)
add_subdirectory(Pico_code/c/lib/LCD)
add_subdirectory(Pico_code/c/lib/Fonts)
add_subdirectory(Pico_code/c/lib/GUI)
add_subdirectory(Pico_code/c/examples)

include_directories(./Pico_code/c/lib/Config)
include_directories(./Pico_code/c/lib/GUI)
include_directories(./Pico_code/c/lib/Icm20948)
include_directories(./Pico_code/c/lib/LCD)
include_directories(./Pico_code/c/lib/Infrared)
include_directories(./Pico_code/c/lib/OLED)
include_directories(./Pico_code/c/examples)

# Add my libraries
add_subdirectory(libs/)
add_subdirectory(myhttpserver/)
include_directories(./libs/)
include_directories(./myhttpserver/)
# End of my libraries

# Enable C++ exceptions
add_compile_options(-fexceptions)

add_executable(first main.cpp)

pico_set_program_name(first "first")
pico_set_program_version(first "0.1")

set(PICO_DEFAULT_USB_BAUD_RATE 115200)

# Modify the below lines to enable/disable output over UART/USB
pico_enable_stdio_uart(first 0)
pico_enable_stdio_usb(first 1)

set(SDK_PATH ${PICO_SDK_PATH})
include_directories(${SDK_PATH}/usr/include)
link_directories(${SDK_PATH}/usr/lib)

# Add the standard include files to the build
target_include_directories(first PRIVATE
    ${CMAKE_CURRENT_LIST_DIR}
    ${PICO_SDK_PATH}
)

# Add any user requested libraries
target_link_libraries(first 
    pico_stdlib
    hardware_spi 
    hardware_i2c 
    hardware_dma
    hardware_pwm
    hardware_pio 
    hardware_interp 
    hardware_timer 
    hardware_watchdog 
    hardware_clocks
    Config
    Infrared
    OLED
    Icm20948
    LCD
    Fonts
    GUI
    examples
    blink
    mywifi
    myhttp
    myhttpserver
)

pico_add_extra_outputs(first)`

`build/build.ninja` was missing as well but now after you updated the vs code extension it seems to work.

will-v-pi, 2024-12-18:
> `build/build.ninja` was missing as well but now after you updated the vs code extension it seems to work.

Is this working for you now then? You didn't make any of the other suggested changes to the CMakeLists.txt, so if it's still not working then you should try making those changes first

will-v-pi, 2025-02-21:
Closing due to no response

closed 2025-02-21 by will-v-pi

qubit999, 2025-02-22:
I'm working with PlatformIO and Arduino Core now. I haven't tested anymore, sorry.