raspberrypi/pico-vscode #136: Extension fails to create or import projects

jnull, 2024-12-15:
Hello everyone,
I wanted to test this extension but run into issues early on. I followed the recommendations in the pdf and installed the extension. Then I tried to create a new project from an example, but the project could not be created. I tried it with the Pico 2, Pico 2W and the Pico 1W to no avail.
I was greeted by the following messages in the dev console for the Pico 2W:
`ERR [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - downloadHelper] Extracting archive file failed: ENOENT: no such file or directory, chmod 'C:\Users\<username>\.pico-sdk\openocd\0.12.0+dev\scripts\target\?1879x?1?.cfg'`
followed by
`ERR [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - NewProjectPanel] Generator Process error: Command failed: "c:\Program Files\Python\Python313\python.exe" "c:/Users/<username>/.vscode/extensions/raspberry-pi.raspberry-pi-pico-0.17.2/scripts/pico_project.py" -board pico2_w -exam -d 0 --project vscode --projectRoot "c:\Users\<username>\IDE\rbPico2W" --sdkVersion 2.1.0 --toolchainVersion 13_3_Rel1 --picotoolVersion 2.1.0 --openOCDVersion 0.12.0+dev --ninjaPath "C:/Users/<username>/.pico-sdk/ninja/v1.12.1/ninja" --cmakePath "C:/Users/<username>/.pico-sdk/cmake/v3.29.9/bin/cmake"  "blink"`
and finally
`ERR [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - NewProjectPanel] Generator Process exited with code: 4294967295`

I am running on Win11Pro x64 Version 24H2 with an up-to-date Vs Code. I also could not find any logs.

Any advise or help is much appreciated.

jnull, 2024-12-15:
I just saw that there is another issue with similar errors: #34 This gave me the idea to switch my language settings. This actually worked. Changing my default language settings from German to English (US) seems to be a work around for the issue.

will-v-pi, 2024-12-17:
I can confirm that I can reproduce this if I change my VS Code display language from English to German, then install the extension and try to create an example project.

Unfortunately, I cannot see what would cause this, so labelling this as a known issue for now

> ERR [Extension Host] [�[31mERROR�[0m] [raspberry-pi-pico - downloadHelper] Extracting archive file failed: ENOENT: no such file or directory, chmod 'C:\Users\<username>\.pico-sdk\openocd\0.12.0+dev\scripts\target\?1879x?1?.cfg'

Note that this error is not actually a problem, it's the Generator Process error that's the actual issue

will-v-pi, 2025-01-09:
I don't seem to be able to reproduce this bug anymore with the latest VS Code version (1.96.2) - @jnull could you see if this bug still occurs for you with the latest VS Code in German (might need to uninstall and reinstall the extension)?

jnull, 2025-01-09:
I reinstalled the extension and tried again with VSS Code version 1.96.2 and System language settings in German it unfortunately did not work. It run into the same issue as before:
```
[Extension Host] [INFO] [raspberry-pi-pico - extension] Extension activation triggered
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - UI] Initializing UI
console.ts:137 [Extension Host] [WARN] [raspberry-pi-pico - extension] No workspace folder or Pico project found.
webviewElement.ts:486 An iframe which has both allow-scripts and allow-same-origin for its sandbox attribute can escape its sandboxing.
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - examplesUtil] Successfully downloaded examples list from the internet.
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - NewProjectPanel] Loaded 118 examples.
console.ts:137 [Extension Host] [raspberry-pi-pico] https://github.com/raspberrypi/pico-examples.git sdk-2.1.0 has been cloned with a sparse-checkout.
console.ts:137 [Extension Host] [raspberry-pi-pico] Sparse-checkout selected example: blink
console.ts:137 [Extension Host] [raspberry-pi-pico] Copying example from C:/Users/<username>/.pico-sdk/examples/blink to c:/Users/<username>/IDE/rbPico2W
console.ts:137 [Extension Host] [raspberry-pi-pico] Done copying example.
console.ts:137 [Extension Host]% 
Object
console.ts:137 [Extension Host] [raspberry-pi-pico] https://github.com/raspberrypi/pico-sdk.git 2.1.0 has been cloned and installed.
log.ts:445   ERR [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - downloadHelper] Extracting archive file failed: ENOENT: no such file or directory, chmod 'C:\Users\<username>\.pico-sdk\openocd\0.12.0+dev\scripts\target\?1879x?1?.cfg'
console.ts:137 [Extension Host] [ERROR] [raspberry-pi-pico - downloadHelper] Extracting archive file failed: ENOENT: no such file or directory, chmod 'C:\Users\<username>\.pico-sdk\openocd\0.12.0+dev\scripts\target\?1879x?1?.cfg'
log.ts:445   ERR [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - NewProjectPanel] Generator Process exited with code: 4294967295
console.ts:137 [Extension Host] [ERROR] [raspberry-pi-pico - NewProjectPanel] Generator Process exited with code: 4294967295
notificationsAlerts.ts:42 Could not create new project: blink
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - NewProjectPanel] 
log.ts:445   ERR [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - NewProjectPanel] Generator Process error: Command failed: "c:\Program Files\Python\Python313\python.exe" "c:/Users/<username>/.vscode/extensions/raspberry-pi.raspberry-pi-pico-0.17.2/scripts/pico_project.py" -board pico2_w -exam -d 0 --project vscode --projectRoot "c:\Users\<username>\IDE\rbPico2W" --sdkVersion 2.1.0 --toolchainVersion 13_3_Rel1 --picotoolVersion 2.1.0 --openOCDVersion 0.12.0+dev --ninjaPath "C:/Users/<username>/.pico-sdk/ninja/v1.12.1/ninja" --cmakePath "C:/Users/<username>/.pico-sdk/cmake/v3.29.9/bin/cmake"  "blink"
console.ts:137 [Extension Host] [ERROR] [raspberry-pi-pico - NewProjectPanel] Generator Process error: Command failed: "c:\Program Files\Python\Python313\python.exe" "c:/Users/<username>/.vscode/extensions/raspberry-pi.raspberry-pi-pico-0.17.2/scripts/pico_project.py" -board pico2_w -exam -d 0 --project vscode --projectRoot "c:\Users\<username>\IDE\rbPico2W" --sdkVersion 2.1.0 --toolchainVersion 13_3_Rel1 --picotoolVersion 2.1.0 --openOCDVersion 0.12.0+dev --ninjaPath "C:/Users/<username>/.pico-sdk/ninja/v1.12.1/ninja" --cmakePath "C:/Users/<username>/.pico-sdk/cmake/v3.29.9/bin/cmake"  "blink"
``` 
It still looks for a wired config: `?1879x?1?.cfg`

will-v-pi, 2025-01-13:
> It still looks for a wired config: `?1879x?1?.cfg`

As I mentioned, ignore that error, the extension already works around it - it’s due to the archive files not being fully compatible with special characters on Windows, and some OpenOCD boards with special characters in the names.

jnull, 2025-01-27:
Just tried creating a project from an example with the latest update and it seems to work. The issue above is still mentioned but it creates the project correctly and opens it afterwards :)

closed 2026-11-05 by will-v-pi