raspberrypi/pico-vscode #138: New C/C++ Project, Import Project and New Project From Example are blank pages

MarcoMagana, 2024-12-18:
1. Installing  VSCodeUserSetup-x64-1.95.3 on Windows 10
2. Installing Raspberry Pi Pico Extension.
3. Click on New C/C++ Project, Import Project, or New Project From Example show blank pages.
[image]

Reinstalling VSCode and Raspberry Pi Pico Extension do not fix the problem.

will-v-pi, 2024-12-18:
The page can sometimes take time to load, but if you close all the pages, close and reopen VS Code, and then try to open one of the pages again (rather than multiple at once) it should work.

Do you get any output in the developer console (Help->Toggle Developer Tools)? Could you also try using the latest debug build which you can download from [Actions](https://github.com/raspberrypi/pico-vscode/actions/workflows/release.yml?query=branch%3Amain), which should provide more output in the developer console?

MarcoMagana, 2024-12-18:
I get the following log:

 WARN Via 'product.json#extensionEnabledApiProposals' extension 'ms-vscode.vscode-selfhost-test-provider' wants API proposal 'attributableCoverage' but that proposal DOES NOT EXIST. Likely, the proposal has been finalized (check 'vscode.d.ts') or was abandoned.
log.ts:435  WARN Via 'product.json#extensionEnabledApiProposals' extension 'ms-python.python' wants API proposal 'contribIssueReporter' but that proposal DOES NOT EXIST. Likely, the proposal has been finalized (check 'vscode.d.ts') or was abandoned.
log.ts:435  WARN Via 'product.json#extensionEnabledApiProposals' extension 'ms-python.debugpy' wants API proposal 'contribIssueReporter' but that proposal DOES NOT EXIST. Likely, the proposal has been finalized (check 'vscode.d.ts') or was abandoned.
log.ts:425  INFO Started local extension host with pid 23336.
log.ts:435  WARN [twxs.cmake]: Cannot register 'cmake.cmakePath'. This property is already registered.
log.ts:445   ERR Ignoring cortex-debug.armToolchainPath.linux as cortex-debug.armToolchainPath is null
log.ts:445   ERR Ignoring cortex-debug.armToolchainPath.osx as cortex-debug.armToolchainPath is null
log.ts:445   ERR Ignoring cortex-debug.armToolchainPath.windows as cortex-debug.armToolchainPath is null
log.ts:445   ERR Ignoring cortex-debug.JLinkGDBServerPath.linux as cortex-debug.JLinkGDBServerPath is null
log.ts:445   ERR Ignoring cortex-debug.JLinkGDBServerPath.osx as cortex-debug.JLinkGDBServerPath is null
log.ts:445   ERR Ignoring cortex-debug.JLinkGDBServerPath.windows as cortex-debug.JLinkGDBServerPath is null
log.ts:445   ERR Ignoring cortex-debug.openocdPath.linux as cortex-debug.openocdPath is null
log.ts:445   ERR Ignoring cortex-debug.openocdPath.osx as cortex-debug.openocdPath is null
log.ts:445   ERR Ignoring cortex-debug.openocdPath.windows as cortex-debug.openocdPath is null
log.ts:445   ERR Ignoring cortex-debug.pyocdPath.linux as cortex-debug.pyocdPath is null
log.ts:445   ERR Ignoring cortex-debug.pyocdPath.osx as cortex-debug.pyocdPath is null
log.ts:445   ERR Ignoring cortex-debug.pyocdPath.windows as cortex-debug.pyocdPath is null
log.ts:445   ERR Ignoring cortex-debug.PEGDBServerPath.linux as cortex-debug.PEGDBServerPath is null
log.ts:445   ERR Ignoring cortex-debug.PEGDBServerPath.osx as cortex-debug.PEGDBServerPath is null
log.ts:445   ERR Ignoring cortex-debug.PEGDBServerPath.windows as cortex-debug.PEGDBServerPath is null
log.ts:445   ERR Ignoring cortex-debug.stutilPath.linux as cortex-debug.stutilPath is null
log.ts:445   ERR Ignoring cortex-debug.stutilPath.osx as cortex-debug.stutilPath is null
log.ts:445   ERR Ignoring cortex-debug.stutilPath.windows as cortex-debug.stutilPath is null
log.ts:445   ERR Ignoring cortex-debug.stlinkPath.linux as cortex-debug.stlinkPath is null
log.ts:445   ERR Ignoring cortex-debug.stlinkPath.osx as cortex-debug.stlinkPath is null
log.ts:445   ERR Ignoring cortex-debug.stlinkPath.windows as cortex-debug.stlinkPath is null
log.ts:445   ERR Ignoring cortex-debug.stm32cubeprogrammer.linux as cortex-debug.stm32cubeprogrammer is null
log.ts:445   ERR Ignoring cortex-debug.stm32cubeprogrammer.osx as cortex-debug.stm32cubeprogrammer is null
log.ts:445   ERR Ignoring cortex-debug.stm32cubeprogrammer.windows as cortex-debug.stm32cubeprogrammer is null
log.ts:425  INFO Settings Sync: Account status changed from uninitialized to unavailable
console.ts:137 [Extension Host] do not activate
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - extension] Extension activation triggered
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - UI] Initializing UI
console.ts:137 [Extension Host] [WARN] [raspberry-pi-pico - extension] No workspace folder or Pico project found.
kYi @ console.ts:137
console.ts:137 [Extension Host]% Object
console.ts:137 [Extension Host]% Object
log.ts:425  INFO [perf] Render performance baseline is 26ms
workbench.html:1 Access to fetch at 'https://www.vscode-unpkg.net/_gallery/marus25/cortex-debug/latest' from origin 'vscode-file://vscode-app' has been blocked by CORS policy: Response to preflight request doesn't pass access control check: It does not have HTTP ok status.
www.vscode-unpkg.net/_gallery/marus25/cortex-debug/latest:1 
        
        
       Failed to load resource: net::ERR_FAILED
log.ts:445   ERR [network] #1: https://www.vscode-unpkg.net/_gallery/marus25/cortex-debug/latest - error GET Failed to fetch
workbench.html:1 Access to fetch at 'https://www.vscode-unpkg.net/_gallery/ms-vscode/cmake-tools/latest' from origin 'vscode-file://vscode-app' has been blocked by CORS policy: Response to preflight request doesn't pass access control check: It does not have HTTP ok status.
www.vscode-unpkg.net/_gallery/ms-vscode/cmake-tools/latest:1 
        
        
       Failed to load resource: net::ERR_FAILED
log.ts:445   ERR [network] #9: https://www.vscode-unpkg.net/_gallery/ms-vscode/cmake-tools/latest - error GET Failed to fetch
workbench.html:1 Access to fetch at 'https://www.vscode-unpkg.net/_gallery/ms-python/vscode-pylance/latest' from origin 'vscode-file://vscode-app' has been blocked by CORS policy: Response to preflight request doesn't pass access control check: It does not have HTTP ok status.
www.vscode-unpkg.net/_gallery/ms-python/vscode-pylance/latest:1 
        
        
       Failed to load resource: net::ERR_FAILED
log.ts:445   ERR [network] #8: https://www.vscode-unpkg.net/_gallery/ms-python/vscode-pylance/latest - error GET Failed to fetch
[... cut ...]
log.ts:445   ERR [network] #25: https://www.vscode-unpkg.net/_gallery/ms-python/debugpy/latest - error GET Fetch timeout: 10000ms
log.ts:445   ERR [network] #28: https://www.vscode-unpkg.net/_gallery/ms-vscode/cmake-tools/latest - error GET Fetch timeout: 10000ms
log.ts:445   ERR [network] #33: https://www.vscode-unpkg.net/_gallery/paulober/pico-w-go/latest - error GET Fetch timeout: 10000ms
log.ts:445   ERR [network] #35: https://www.vscode-unpkg.net/_gallery/twxs/cmake/latest - error GET Fetch timeout: 10000ms

will-v-pi, 2024-12-18:
That looks like you’re having some internet problems not related to this extension, given its printing issues with other extensions too - possibly your anti-virus or firewall causing issues?

MarcoMagana, 2025-01-07:
It looks like some firewall policies do not let me access some sites required to develop with the Raspberry Pi Pico plugin. I've tried at a different location (SSID) and had no problem. But, I cannot create projects either when not connected to the Internet. So, how do I overcome that when I'm on the move? Is there a way to configure the plugin to work offline?

will-v-pi, 2025-01-07:
You should be able to create projects when not connected to the internet, provided you've previously created a project with the same settings while connected to the internet, so that the Toolchain, SDK, CMake etc have been downloaded.

I've confirmed that this works for me, so I can create new projects when offline, provided I've previously created a project while online in order to download all the tools.

MarcoMagana, 2025-01-10:
I created a project from an example, several of them actually and I always get the pop-up message "Unable to find compilation information for this file"
[image]

will-v-pi, 2025-01-14:
Are you clicking any buttons to get that error to pop up, or does it just pop up as soon as you open the file? That error isn't coming from our extension, so could you try and find out what other extension is throwing it?

My guess would be that you're trying to use a different extension to compile the file, which is then throwing that error - to use our extension you should click the "Compile" button in the bottom right, next to the "Run" and "Pico SDK" buttons.

MarcoMagana, 2025-01-14:
I got it. Thanks.

I can now generate code examples and new projects, compile, and copy the uf2 file in BOOTSEL mode to run the program.

closed 2025-01-14 by MarcoMagana