raspberrypi/pico-vscode #140: MicroPython cannot find the module.

QQ2046413092, 2024-12-21:
Hello, I am a beginner here. Can I ask some basic questions? I am using the micropython plugin to write code, and I have created a new project. However, I am facing a problem where modules located in the same directory as the project cannot be imported. When I run the code, it gives an error saying that the module cannot be found.

QQ2046413092, 2024-12-21:
There is another issue where, when running a MicroPython project and disconnecting the circuit board suddenly, the "Run" button in the status bar of Visual Studio Code will become stuck and will not stop, even after trying to reconnect the circuit board and stopping. The only solution is to restart Visual Studio Code or the extension.

frohro, 2024-12-23:
I don't know if this is the same bug, but if I start a new MicroPython project, and let it suggest the Blink project, it cannot find the module machine.  I can copy the Blink project into Thonny and it runs beautifully.

paulober, 2024-12-23:
> Hello, I am a beginner here. Can I ask some basic questions? I am using the micropython plugin to write code, and I have created a new project. However, I am facing a problem where modules located [...]

Thank you for reaching out! The issue you're experiencing occurs because the modules are located on your host machine and not on the Pico itself. When you execute a script on the Pico, its Python interpreter cannot resolve or access those modules unless they are uploaded to the device.

To resolve this, I recommend uploading either the entire project or just the specific modules you need to the Pico before running your script. Keep in mind that if you update any of the modules locally, you will need to upload the changes again to test them on the Pico.

paulober, 2024-12-23:
> I don't know if this is the same bug, but if I start a new MicroPython project, and let it suggest the Blink project, it cannot find the module machine. I can copy the Blink project into Thonny and it runs beautifully.

How do you execute the blink script? (Keep in mind that the default run button does not work if you want to execute the script on the connected Pico. Instead I recommend the run button contributed by the extension in the toolbar, the context menu or as command in the command palette)

frohro, 2024-12-24:
@paulober you solved my issue.  I was using the regular "Run" command at the top.  The one in the toolbar works fine!

allaire, 2025-01-19:
@paulober This thread helped, thanks. 

Could you please explain me how the run command works exactly? Does it uploads the current open file every time I press run on the Pico?

When I press run, the code executes correctly on my pico, then I unplug it and run it on power source, but then pico does not contains my last changes from the last run. I have to plug it again and choose "upload project to pico".

I haven't found anything about that in the doc

paulober, 2025-01-20:
@allaire Of course. The run button/command does copy and execute the contents of the currently open file but it doesn't store it. Another option would be do a right click in the editor and select `Upload file to Pico` if you want to copy the latest changes of this one file.

allaire, 2025-01-20:
Brilliant thank you so much.

closed 2026-11-05 by will-v-pi