raspberrypi/pico-vscode #141: Toolchain install failures in Windows 11

Funkenjaeger, 2024-12-28:
I've just been through about 4-6 hours of struggle to install this extension (in latest vscode, in Win11), create new project from example 'blink', compile and run.  
Initially I had many other extensions installed in vscode, as I use it for other projects, but I tried using a fresh profile, and ultimately wiped vscode completely and started fresh, to no avail.

The failure mode was that upon initiating "new project from example" from the pico extension, I begin getting the notifications in the bottom right about it installing the toolchain, but it would make some progress and then fail - sometimes ending with an error that it failed to install the toolchain, sometimes an error that it failed to install CMake.  At that point the project folder existed, however the 'build' folder was empty; hitting 'compile' just errored out as build.ninja didn't exist, and the commands 'configure cmake' and 'clean cmake' in the pico extension both error out.  Closing and reopening vscode didn't cause it to reattempt the toolchain install.  Going as far as uninstalling all extensions and vscode itself, deleting c:\users\<username>\.vscode and %APPDATA%\Code, reinstalling all from scratch, which also didn't cause it to reattempt the toolchain install.

After some trial and error I found that closing vscode, deleting the c:\users\<username>\.pico-sdk directory, reopening vscode and attempting to create the example project again was sufficient to trigger it to reattempt toolchain installation.  Amidst my googling, I found one lonely forum post where someone said to just keep trying the toolchain install repeatedly until it works, because it's fragile and doesn't recover from any sort of hiccups...  So I just repeated this again and again - and frustratingly, it would fail with different errors on successive tries of exactly the same process - until a half dozen or so tries later, it finally completed the entire toolchain install without errors, the build folder of the blink project was not empty, and it compiled and ran fine.
Needless to say, while I'm glad it finally worked, this "definition of insanity" approach was very frustrating.

I'm at a loss as to what exactly was failing in these instances, since I was unable to find any relevant log messages in any of the vscode outputs, and don't know where to look for log files (if there are any) - all I really got was the vscode notifications, which are very terse.

Ideally it would be great if the installation process could gracefully handle such errors, but failing that, it would be helpful if it could at least:
A) provide some useful log message outputs to the user during toolchain install, especially in the event of failures, and 
B) have some ability to recover from a failed install, even if it's as basic as the nuclear approach of wiping the .pico-sdk directory in the event of unrecoverable toolchain install errors

will-v-pi, 2025-01-06:
A) Unfortunately we don't have any more useful information to give when the toolchain install fails - it's usually just due to a poor internet connection. You can also manually download the toolchain if you're have issues, see https://github.com/raspberrypi/pico-vscode/pull/42

B) If the toolchain download fails then it shouldn't create anything in the .pico-sdk directory, so re-attempting the download should work. Once it's reached the stage where it downloads CMake (as in your original test), it has successfully installed the toolchain, so you should be able to just run it again and it'll try to install CMake again. If it has gone wrong, then you can just delete the .pico-sdk directory and try again, as you describe. We don't have a way to detect an installation that has gone wrong in this way (as if it's created files in .pico-sdk but not completed successfully, we don't know which files are missing), so this needs to be a manual troubleshooting step.

Funkenjaeger, 2025-01-15:
My experience was that it always created files in the .pico-sdk directory, even when it told me the toolchain install failed.  On a few occasions re-running would let it get slightly further (such as, on the first try it says toolchain install failed, on the next try it said cmake install failed).  But mostly, re-running didn't cause it to try installation again, it would just stay in the final broken state described above indefinitely - until I deleted the .pico-sdk directory to force an actual fresh start.

Anyway, as it's clear that a fix to automatically handle these types of issues is not practical, I'll close this.

closed 2025-01-15 by Funkenjaeger