raspberrypi/pico-vscode #142: C/C++ project fails

Timfedora, 2025-01-01:
I have installed the beta version 17.2 and encountered an issue when creating a new C project. Although the project creation process initiates, it never generates any files or output. Instead, the process closes without any result or error messages.

Here are the steps I’ve taken so far to resolve the issue:

Reinstalled the beta version 17.2.
Reinstalled Visual Studio Code (current version: 1.69.2).
 
Please let me know if further information is needed to troubleshoot this issue.

Timfedora, 2025-01-01:
Here is what the issue looks like.

https://github.com/user-attachments/assets/4b5e959e-2e98-4c65-86f7-32f2d3e1aada

And here is some other info.

python --version
Python 3.13.1

tar --version
tar (GNU tar) 1.35
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Written by John Gilmore and Jay Fenlason.

git --version
git version 2.47.1

gcc --version
gcc (GCC) 14.2.1 20240912 (Red Hat 14.2.1-3)
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

cmake --version
cmake version 3.30.5
CMake suite maintained and supported by Kitware (kitware.com/cmake).

make --version
GNU Make 4.4.1
Built for x86_64-redhat-linux-gnu
Copyright (C) 1988-2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

If there is any thing else you need just let me know.

will-v-pi, 2025-01-06:
When you get the oddly coloured UI at 0:06, this is a bug - you should close that UI then reopen it so that you get the correctly coloured UI you see at 1:04. Also, it can take some time to clone the pico examples the first time (which has no progress bar), so you may need to wait slightly longer after clicking Create than you are currently - but I think the main issue is the bug of the UI not loading correctly, which then causes the extension to not work correctly.

firew4lker, 2025-01-07:
The latest debug vsix fails to open with:

Cannot activate the 'Raspberry Pi Pico' extension because it depends on an unknown 'ms-vscode.cpptools' extension.

I am attaching the log file for the current version.

[image]

[vscode-app-1736262814627.log.txt](https://github.com/user-attachments/files/18334579/vscode-app-1736262814627.log.txt)

will-v-pi, 2025-01-07:
@firew4lker Are you using VS Codium (not VS Code)? If using VS Codium, you need to install the ovsx .vsix file, as the VS Code ones contain extension dependencies that aren't available on VS Codium

Timfedora, 2025-01-08:
Here's what it looks like. 
[image]
[con.txt](https://github.com/user-attachments/files/18339865/con.txt)

Timfedora, 2025-01-08:
[image]

will-v-pi, 2025-01-08:
It looks like you may have cancelled the installation of the examples at some point, leading to a bad directory structure. Can you confirm if there is a `/home/tim/.pico-sdk/examples/.git` directory present - if there is you should just be able to delete the `/home/tim/.pico-sdk/examples` directory and retry the extension, and it should work (just wait longer after clicking Create).

Timfedora, 2025-01-08:
Ok

Timfedora, 2025-01-17:
Ok thanks for your help I got everything working

will-v-pi, 2025-02-21:
Thanks, closing this issue now you've got it working

closed 2025-02-21 by will-v-pi