raspberrypi/pico-vscode #144: File Explorer opens after device reboot

John-Forsyth, 2025-01-04:
I'm pretty new to this and am trying to follow the blink tutorial in the getting started guide. I've got to the point where I click run and see `The device was rebooted to start the application.` in the terminal, but all that happens is file explorer reopens to show me the pico as a mass storage device -- nothing from the LED. 

If anyone could point me in the right direction, that'd be great :)

pico info:
UF2 Bootloader v3.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2

vs code version: 1.96.2

windows:
OS Name	Microsoft Windows 10 Home
Version	10.0.19045 Build 19045

installation: 
Identifier
raspberry-pi.raspberry-pi-pico
Version
0.17.2

will-v-pi, 2025-01-06:
Did you select the correct board type (in this case a Pico, not the default Pico 2)?

will-v-pi, 2025-02-21:
Closing this, as I believe the issue was selecting the wrong board

closed 2025-02-21 by will-v-pi