raspberrypi/pico-vscode #145: Adding Packages from PYPI to Pico board

Netexcavator, 2025-01-06:
I realized when working with the extension that there is no way to import the library from something like PYPI like you can with Thonny. I did see on the MicroPico extension that there was a way to add MIP packages but none of the current MIP packages would work for what I am working on. Is there a way to do it currently or will it be added in the future?

will-v-pi, 2025-01-07:
Currently you can do it manually by downloading the python files from the package you want and uploading them to the device, usually to the path `/lib/<packageName>/__init__.py`

Note that many libraries on PyPi will not work with MicroPython, as they require modules that MicroPython does not have (eg numpy, atexit)

will-v-pi, 2025-04-15:
Closing as not planned, as PyPi packages are not guaranteed to work, and you can download them manually if you want to try them

closed 2025-04-15 by will-v-pi