raspberrypi/pico-vscode #148: Failure to install tools when creating/importing project

infovore, 2025-01-08:
**Describe the bug**
After a fresh install of the Pico VS Code extension, creating (whether bare or from example), or importing a project, fails. The failure is that the toolchain fails to install correctly.

**Extension Logs**
The failure is fetching `pico-sdk-tools`; the `~/.pico-sdk` directory has the SDK and toolchain directories already populated. Developer console makes this clear:

```
[raspberry-pi-pico - githubREST] Fetching pico-sdk-tools releases failed: Error http status code: 404
```

**To Reproduce**
Create a new project in the Extension, eg, from example, Blink; the tools install fails, and cannot be re-started without deleting the `~/.pico-sdk` directory. (See other issues for clarifcation on why the latter happens).

**Expected behavior**
The toolchain should install correctly.

**Platform (please complete the following information):**
Mac OS 14.7, arm64

will-v-pi, 2025-01-08:
I can't reproduce this - on my Arm64 Mac (running MacOS 15.2) creating the blink example project works fine.

Could you see if you can manually access the GitHub API by going to https://api.github.com/repos/raspberrypi/pico-sdk-tools/releases/tags/v2.1.0-0 - if you get a 404 error there as well then this is likely a firewall/internet connectivity problem on your end.

Also, could you post the rest of your logs, because that failure should not cause a fatal error, as the extension has backup ways of downloading the tools when the GitHub API is inaccessible (usually due to rate limiting).

infovore, 2025-01-08:
@will-v-pi very strange: I can see the API just fine from within Chrome or via `curl`. But via VS Code:

[image]

Although: I have not set a `githubToken`, and it could be that the request is being reject for being unauthed. (Although curl, from my command line, is also un-authed.)

infovore, 2025-01-08:
Update: I created a token and used that, and the same error. I'm going to try from a different network shortly.

will-v-pi, 2025-01-08:
It looks like you have a version mismatch - are you definitely using the latest version of the extension (0.17.2)?

From your logs, it's cloning the 2.0.0 examples but the 2.1.0 SDK, which suggests you're using an older extension version that doesn't support SDK 2.1.0 (which would also explain the 404 error)

infovore, 2025-01-08:
aha - I have 0.16.1, which is what I got from the marketplace. Let me check latest.

infovore, 2025-01-08:
Oh god. Apologies for wasting your time:

- VS Code had managed to put itself in quarantine...
- ...meaning it couldn't auto-update
- ...meaning the latest versions it was getting from marketplace was 0.16.1.

Some messing around with `xattr` and some updates later, and all is fine. I'm closing this and apologising for wasting your time. Looking forward to exploring the extension - been using the CMake/CortexDebug plugins on their until now, but making it easier ot install is always good.

closed 2025-01-08 by infovore