raspberrypi/pico-vscode #153: Better defaults for "debug project"

infovore, 2025-01-13:
Not a bug, but a suggestion / feature request:

- The default compilation flag in CMake is, I believe, `Release`; it'd be nice if when you chose to Debug a project, it compiled in `Debug` mode (which I'm not sure it does).
- Similarly, it'd be great if in Debug mode, the compiler flags `-g -O0` were added; as it is, when I debug a project, the debugger works... but all my variables are optimized out and it's very hard to track anything.

My knowledge of Cortex Debug is... thin, but I've had it working on previous RP2040 projects, and in general I'm a _huge_ fan of making this stuff available to end users.

As it is, I have these lines in my `CmakeLists.txt`:

```
set(CMAKE_BUILD_TYPE "Debug")
set(CMAKE_C_FLAGS_DEBUG "-g -O0")
set(CMAKE_CXX_FLAGS_DEBUG "-g -O0")
```

and I comment out the first for release building.

infovore, 2025-01-13:
That said - these issues _might_ all be my fault in that a) I already had a CMake/Cortex Debug configuration setup and b) I'm actually using a custom board type, in a `board` subdirectory in [my project](https://github.com/16n-faderbank/16next_firmware/), and I note in `launchTargetPath.mts` that there's a check for boardtype in there that might affect this. Previous experience of Cortex Debug have been... similarly fragile, and I know things are in beta, but perhaps there's something here.

will-v-pi, 2025-01-14:
To set the de-optimised debug flags, you can use `set(PICO_DEOPTIMIZED_DEBUG 1)` instead of setting the `CMAKE_C..._FLAGS_DEBUG` directly - the default Debug build of the SDK still has some optimisations, which `PICO_DEOPTIMIZED_DEBUG` disables. We wouldn't add this by default, but could add a code generation option to enable it.

I think we could add the automatic switch to Debug builds when using Cortex-Debug, vs release builds when you click Run - it would just require checking the build type and possibly running a CMake Configure whenever you switch between Cortex-Debug and Run.

infovore, 2025-01-23:
Good to know about that flag. I like the idea of switching to Debug builds automatically... but understand that there are good reasons to make that switch manual / use a debugger with a Release build. It's just it turns out a lot of the functionality of the debugger for me was actually specific to deoptimized code.

I'm not sure what the right answer is here, hence being discursive; I _do_ know that the Debug Probe is an excellent thing, and helping more users use it more easily is a win for everybody, so I'll keep seeing how I get on here. Thanks for your patience with me!

will-v-pi, 2025-01-23:
I've added a `switchBuildType` command, which allows for manual switching between build types - I'm currently thinking that automatic switching would be quite annoying for me (and probably others), so I think for now manual switching is probably best. Although we might add it as a feature that you can enable through a setting, but I think the default should be disabled.

IvanVeloz, 2025-01-24:
I'm on the same boat here. I want Cmake to have a Debug configuration when debugging, so that I can add compile flags like `-DLOGGING_LEVEL_DEBUG=1`.

will-v-pi, 2025-04-15:
Closing this as I don't think automatic switching would be desirable, and you can configure any CMake variable you want in CMakeLists.txt to define the flags you want for a given build type

closed 2025-04-15 by will-v-pi