raspberrypi/pico-vscode #156: Visual Studio Raspberry Pi Pico Extension fails to download SDK, claims Python3 is not installed and in path, except that it is.

glenbarney, 2025-01-18:
**Describe the bug**
Visual Studio Raspberry Pi Pico Extension fails to download SDK, claiming incorrectly that Python3 is not installed or in the Path.

On a clean, brand new deployment of Visual Studio Code (with all profiles wiped and all extensions removed), on a computer with Windows 11 Pro and Python 3.13.1 correctly installed and fully functional, the Raspberry Pi Pico extension fails to download the SDK.  Error messages seen are:

Failed to download and install SDK. Make sure all requirements are met.  
Python3 is not installed and in path

**Extension Logs**
[Extension Host] do not activate
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - extension] Extension activation triggered
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - UI] Initializing UI
console.ts:137 [Extension Host] [WARN] [raspberry-pi-pico - extension] No workspace folder or Pico project found.
index.html:283 No service worker controller found. Waiting for controllerchange.
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - examplesUtil] Successfully downloaded examples list from the internet.
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - NewProjectPanel] Loaded 118 examples.
tailwindcss-3_3_5.js:62 cdn.tailwindcss.com should not be used in production. To use Tailwind CSS in production, install it as a PostCSS plugin or use the Tailwind CLI: https://tailwindcss.com/docs/installation
main.js:391 [raspberry-pi-pico] Set theme mode to: dark
console.ts:137 [Extension Host] [raspberry-pi-pico] https://github.com/raspberrypi/pico-examples.git sdk-2.1.0 has been cloned with a sparse-checkout.
console.ts:137 [Extension Host] [raspberry-pi-pico] Sparse-checkout selected example: blink
console.ts:137 [Extension Host] [raspberry-pi-pico] Copying example from C:/Users/Glen/.pico-sdk/examples/blink to c:/Users/Glen/Desktop
console.ts:137 [Extension Host] [raspberry-pi-pico] Done copying example.
console.ts:137 [Extension Host]% 
{}
console.ts:137 [Extension Host] [raspberry-pi-pico] https://github.com/raspberrypi/pico-sdk.git 2.1.0 has been cloned and installed.
log.ts:445   ERR [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - download] Python3 is not installed and could not be downloaded.
console.ts:137 [Extension Host] [ERROR] [raspberry-pi-pico - download] Python3 is not installed and could not be downloaded.
notificationsAlerts.ts:42 Python3 is not installed and in PATH.
log.ts:445   ERR [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - NewProjectPanel] Failed to download and install SDK.
console.ts:137 [Extension Host] [ERROR] [raspberry-pi-pico - NewProjectPanel] Failed to download and install SDK.
notificationsAlerts.ts:42 Failed to download and install SDK. Make sure all requirements are met.
﻿
**To Reproduce**
1, Windows 11 Pro, all updates applied
2. Latest Version of Python (3.13.1) installed from Store, confirmed working
3. Fresh install of Visual Studio Code (when repeating, removed all extensions and profiles first)
4. Install Raspberry Pi Pico extension.
5. Click on extension Icon
6. "New Project From Example"
7. Example: Blink
8. Board type: Pico W
9. Location: Desktop
10. Click Create Button
11. Observe error messages in lower right corner of screen:

Failed to download and install SDK. Make sure all requirements are met.  
Python3 is not installed and in path

[image]

**Expected behavior**
Per Raspberry Pi Getting Started Guide ( https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf ) , exepected behavior is "The extension will now download the SDK and the toolchain, install them locally, and generate the new project. "

**Screenshots**
Screenshot Attached.

[image]

**Platform (please complete the following information):**
 - OS: Windows 11 Pro, 
 - Architecture: X86_64
 - Python 3.13.1 
 - Visual Studio Code 1.96.4 (User)
 - Extension: raspberry-pi.raspberry-pi-pico Version 0.17.2 Published 2024-03-27, 09:58:08  Last Released 2024-11-25, 03:58:18

**Additional context**
1. Python is in fact installed and working
2. Visual Studio code can build and run regular python files and projects
3. In Settings, Raspberry Pi Pico Python 3 path IS set (was set automatically) and is correct:  c:\Users\Glen\AppData\Local\Microsoft\WindowsApps\python3.13.exe
4. That file does exist:

C:\Users\Glen\AppData\Local\Microsoft\WindowsApps>python3.13
Python 3.13.1 (tags/v3.13.1:0671451, Dec  3 2024, 19:06:28) [MSC v.1942 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>

5. If I create a folder "test" and a one-line "hello.py" file in that folder in Visual Studio Code, an attempt to "Run Without Debugging" succeeds:

PS C:\Users\Glen\Desktop\test>  c:; cd 'c:\Users\Glen\Desktop\test'; & 'c:\Users\Glen\AppData\Local\Microsoft\WindowsApps\python3.13.exe' 'c:\Users\Glen\.vscode\extensions\ms-python.debugpy-2024.14.0-win32-x64\bundled\libs\debugpy\adapter/../..\debugpy\launcher' '55830' '--' 'c:\Users\Glen\Desktop\test\hello.py' 
Hello world!

A manual run from the same Terminal window also works:

PS C:\Users\Glen\Desktop\test> python hello.py  
Hello world!

PS C:\Users\Glen\Desktop\test> python3.13 hello.py
Hello world!

6. Others are complaining about similar issues:
https://forums.raspberrypi.com/viewtopic.php?t=376050
https://forums.raspberrypi.com/viewtopic.php?t=376007

7. It has been suggested that changing the Python Locator from "native" back to "js" might help.  It does not.

will-v-pi, 2025-01-20:
This might be fixed by #155 - note that the Python handling has been completely revamped since those forum issues were created, so they will likely be unrelated to this and may have been fixed.

I'm afraid that like the other similar python issue in #127, I'm unable to reproduce this on my system (following your exact reproduction steps), and therefore can't figure out what is going wrong.

I think the simplest workaround would be to uninstall Python, clear the VS code settings, then follow the steps again without Python installed and the extension will download it's own copy of Python to use. You can then re-install Python from the Microsoft store to use with the rest of your system.

glenbarney, 2025-01-23:
Thank you for this.  I am happy to confirm that this workaround got me up and running.  I did also have to install the ZADIG drivers to talk to the Pico board, but that's not related to this issue, just mentioning it in case someone else trips over it.

WORKAROUND:

Starting with a fresh reset of VS Code AND removing preinstalled Python permitted the extension to install and run properly.  Following that, I reinstalled python and all of my other VS code extensions, and confirmed that I could still create, compile and run new PICO projects.

Because you cannot duplicate this and because I feel that wiping out existing Visual Studio Code configurations is not acceptable just to install and run this extension, I tried to do some additional testing to assist.  Here is the narrative for what I tried::

ADDITIONAL TESTING:  BRACKETING

Looking at Issue #155 that you referred to, I then reset Code again, wiped everything, but did NOT uninstall Python3 from the store.  I then launched Code and installed the extension.  

Immediately after installing I looked at settings, and I see that the Python3 Path is empty.  I then attempted to create a new project again, as in my original trouble report.  Unfortunately it failed.  I got this:

console.ts:137 [Extension Host] [ERROR] [raspberry-pi-pico - download] Python3 is not installed and could not be downloaded.
notificationsAlerts.ts:42 Python3 is not installed and in PATH.

ADDITIONAL TESTING: CLEAR PYTHONPATH

I looked at settings again, and see that it had been set to c:\Users\Glen\AppData\Local\Microsoft\WindowsApps\python3.13.exe - which, again, IS valid and IS present.  I erased the field at that point.  

I then tried to create the project again.  It sat running for 3-4 minutes, and eventually built the project, but could not compile it.  There were a slew of errors but it came down to:

C:/Users/Glen/Desktop/blink/blink.c:12:10: fatal error: pico/cyw43_arch.h: No such file or directory

I noted that the PythonPath had been repopulated again.

ADDITIONAL TESTING: CLEAR PYTHONPATH, WIPE SDK

So this time I wiped that field out, wiped the project out, AND removed the downloaded SDK in \Users\glen\.pico-sdk, and restarted Code.

That failed.  Pythonpath got repopulated with Microsoft Store Python, and the SDK download failed.

ADDITIONAL TESTING: JUST UNINSTALL PYTHON

I tried just uninstalling Python, and retrying the project create.  That failed with the spinning "Reactivating Terminals" message.

ADDITIONAL TESTING: CLEAR PYTHONPATH (With Python uninstalled)

I then tried clearing the pythonpath with python uninstalled, then making a new project.  That also failed with "Reactivating Terminals" spinning forever.

When I see that error, the console says this:

mainThreadExtensionService.ts:78 [ms-python.python]Failed to resolve env "C:\\Users\\Glen\\AppData\\Local\\Microsoft\\WindowsApps\\python3.13.exe"

I then went searching through all the settings to see if the leftover python path was showing up anywhere else - it was not.

ADDITIONAL TESTING: REMOVE SDK

The pythonpath in settings was still blank, despite the error message in the console.  So I then tried removing the SDK.  That ALSO failed with the "Reactivating Terminals" loop.

ADDITIONAL TESTING: REMOVE SDK AND EXTENSION

I then tried uninstalling the Raspberry Pi Pico extension, MicroPico, CortextDebug, RTOS Views, and Memory View.  Then wiped the SDK again, restart Code, and reinstall the extension(s).

That ALSO failed with the "Reactivating Terminals" loop.

ADDITIONAL TESTING: WIPE APPDATA\ROAMING\CODE (but leave extensions and .pico_sdk installed)

This seems to have triggered a redownload of the SDK prior to the building of the project.  A compile at this point succeeded.

FURTHER SEARCHING

Having apparently narrowed this to a problem in the profile, I then reset everything to my previous state.  I wiped pico_sdk, the code profile, and all extensions.  I reinstalled Python from the Store.  I redownloaded the PIco SDK extension and let it fail.  I cleared the pythonpath in the setting.  

I then did a search for any other references to python in the entire AppData/Roaming/Code subtree.

Glen@HPZ6 /cygdrive/C/Users/Glen/AppData/Roaming/Code
$ find . -type f -exec grep -il 'python3.13' {} \;
./logs/20250122T185417/window1/exthost/ms-python.python/Python Locator.log
[... 50 lines cut ...]
      }
    ]
  ]
}

So I went in to AppData\Roaming\Code\User\settings.json and removed the PythonPath from there, and then I went to AppData\Roaming\Code\User\globalStorage and removed ms-python.python

Unfortunately, that didn't help, because as soon as I started VSCode I got the "Reactivating Terminals" spinner again.

I had to also remove state.vscdb and state.vscdb.backup... which of course is probably a non-trivial step.

CONCLUSION

Your workaround did work, thank you for that.  That said, I feel that this is not a solution, but, as you say, a workaround.

I don't pretend to understand the nuances of what's happening in #155 but I think it's not enough to blank things out in the extension settings itself.  I think the problem here is that VS Code is still attempting to use a user-installed version of Python even if the extension python path is cleared.  I also think that even removing the user-installed Python isn't enough, because VSCode seems to "remember" the previously-installed Python and get confused by its absence ("Reactivating Terminals").  

[... cut ...]
At this moment, I can reproduce this on other Windows 11 laptops.  The trick to breaking it is just to have Python 3.13 installed from the Store before you first download the Pico extension.  The download will fail.  In my case I had a fully-customized, active VS Code environment, and the only way I could get the Pico extension to download the SDK and work properly was to not only uninstall Store python, but also wipe out all of my settings across the entire Code database.  

I suggest that that is a big ask for people, and I hope this information helps inform a potential solution.  I hope this information is helpful.

Thanks!
Glen

will-v-pi, 2025-01-23:
Given this and #127 seem to be issues with the Windows Store python (which is a reparse point), I think I'll just add a check in the extension to ignore the Windows Store python. It'll then revert to downloading it's own python, which should work better.

The "Reactivating Terminals" thing will be unrelated to this extension, so I don't think we can fix that - you might have something in your environment that depends on python to activate the terminal, or something like that.

closed 2025-01-23 by will-v-pi