raspberrypi/pico-vscode #159: Improve multiple OS support

rafaelcorsi, 2025-01-31:
Hello, I am a professor of an embedded systems course in computer engineering for undergraduates, and I was happy with this extension, this saves time in the classroom. However, the way it handles multiple OS does not make it easy to use in this scenario.

The extension does not work well when used directly (just open the project folder) on an OS different from the one where the project was created. From what I observed, it uses relative paths that are OS-specific. I understand that the current solution is to "import" the project, but I didn't find this to be a smooth workflow for classroom use. Ideally, a student should be able to clone the repository, open the folder in VS Code, and start using it right away. However, to use it with the extension, there is an extra step of updating files in the .vscode folder.

Issues I see with this:
- Many assessments are done in groups, meaning each student will have to import the project locally and remember not to add the .vscode folder to Git (yes, we can use a .gitignore file).
- Students use multiple devices, often with different OSes, and they won’t always remember to import the project.

A solution that doesn’t require much restructuring would be for the extension to detect when the OS is different from the one the project was configured for and notify the user that an import is necessary, automatically displaying a popup to update the configuration.

will-v-pi, 2025-02-01:
We have actually worked to make sure that it is easy to open a project created on one OS on a different one - we have tested this across Windows, MacOS, and Raspberry Pi OS without issue. Could you clarify what paths you’re seeing that are OS specific?

Note that projects created with older versions of the extension may have device specific paths (eg the Python path), but this has been fixed since version 0.17.0 - you can just delete the device specific Python path setting from older settings.json files.

> remember not to add the .vscode folder to Git (yes, we can use a .gitignore file).

You should commit the .vscode folder to Git when using this extension - this may be what’s causing your issues. That folder should only contain project specific settings, and any device specific settings should be in the user’s settings.json file instead. If you don’t commit that folder then you will need to import the project when you clone it from Git, as you’re observing.

We could improve the default .gitignore to ensure the .vscode folder is committed by adding `!.vscode/*` to it.

rafaelcorsi, 2025-02-01:
@will-v-pi I will test it again and return to you, thank you very much for your attention.

commit 57e43c27 (2025-02-24): Explicitly don't ignore .vscode in default .gitignore

Fixes #159