raspberrypi/pico-vscode #162: pico-vscode breaks when VS Code on PI OS upgrades to 1.97.1

GitGlenn, 2025-02-13:
_Before filing a bug report, please try to reproduce your issue using the latest debug .vsix file from [GitHub Actions](https://github.com/raspberrypi/pico-vscode/actions/workflows/release.yml?query=branch%3Amain). This will provide more debug information than the release build._

**Describe the bug**
A clear and concise description of what the bug is.

**Extension Logs**
Relevant extracts of the extension log showing the errors with this extension, which can be viewed using Help -> Toggle Developer Tools and clicking on the Console.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - OS: [e.g. Windows 11, macOS, Raspberry Pi OS]
 - Architecture: [e.g. x86, Arm64]

**Additional context**
Add any other context about the problem here.

paulober, 2025-02-13:
@GitGlenn Please fill out the template.

GitGlenn, 2025-02-13:
VS code updated automatically to 1.97.1
Version: 1.97.1
Commit: e249dada235c2083c83813bd65b7f4707fb97b76
Date: 2025-02-10T13:13:58.153Z
Electron: 32.2.7
ElectronBuildId: 10660205
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Linux arm64 6.6.74+rpt-rpi-2712

[image]

Here are links to photos of the error message and effect on Raspberry PI 5

https://photos.app.goo.gl/ZXUq2N7CfezAX3cb8

Please let me know what logs or other information is useful and where the templates are and how to attach them.

Thank you, (sorry a newbee at posting in forums) Glenn

GitGlenn, 2025-02-14:
Hi Paul,
I left a comment with some updated info. If I can provide more, please let me know what I can provide and where I should provide.
I have to admit, the pico-vscode was working great until two days ago when VScode automatically to 1.97.1 and now I cannot even do a compile. 
Thanks, Glenn

GitGlenn, 2025-02-14:
Upgrade to version 1.97.2 (at least 1.97.1 is flawed)

GitGlenn, 2025-02-14:
Hi Paul,
Found there was an update release available today. 1.97.2 fixes the problem I (tried to) list.

will-v-pi, 2025-02-14:
Looks like this was an upstream issue with VS Code https://github.com/microsoft/vscode/issues/239838 - closing as it has now been fixed.

closed 2025-02-14 by will-v-pi