raspberrypi/pico-vscode #165: Support Windows 11 ARM

pete4radio, 2025-02-19:
Please add Windows 11 ARM support.  The extension's splash page says only x86_64 is supported, and indeed, VSCode says "There is no data provider registered..." when VSCode loads the Raspberry Pi Pico Extension.

will-v-pi, 2025-02-19:
There are two prerequisites with other tools before we can support this:
- The Serial Monitor extension needs Arm64 support added https://github.com/microsoft/vscode-serial-monitor/issues/148
- There needs to be an Arm toolchain which supports Windows on Arm (see https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads)

For now, you can use the workaround of downloading the x86_64 version of VS Code and using the extension with that, rather than the native Arm64 version - this should run fine using the x86_64 emulation built into Windows on Arm.

pete4radio, 2025-02-20:
The work-around is very useful, thanks for mentioning that.  It runs the x64 ARM toolchain (under the x64 VS Code).  Maybe the ARM RPi extension could also run that toolchain.

I wonder if whatever is tripping on the ARM serial monitor could either be disabled or use the x64 version too.  Alternatively, the serial monitor is very convenient but the user could be informed about screen or PUTTY.

Perhaps the least satisfactory solution is to let ARM VScode load the ARM RPi Extension even tho it will fail without a useful diagnostic message.  I spent many hours trying to figure out what the problem was.

closed 2026-04-01 by will-v-pi