raspberrypi/pico-vscode #166: On Windows 11 keep receiving warning: could not convert 'main' from the host encoding (CP1252) to UTF-32. This normally should not happen, please file a bug report when debugging. Can step through code but only other functions.

SagaraBattousai, 2025-02-22:
**Describe the bug**
With versions 0.16.2,  0.17.4 and 0.17.5 when debugging with cortex-debug I keep reciving this warning message: warning: could not convert 'main' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.

I can step through the code but only for functions other than main. I have tried changing the file encoding, activating the beta in Windows command prompt to encode all as utf-8 and manually setting the path to the arm-none-eabi gcc.

will-v-pi, 2025-02-24:
I can't reproduce this - could you just confirm that you still see this if you create a New Project From Example using the blink example, and send screenshots/video showing exactly what's happening?

Also, note that if your code is optimised, then it may look like it's starting in a different place other than main (eg with the blink example it will actually pause in `pico_led_init` rather than at the start of `main`) - this is not a bug, it's due to function inlining the compiler does when optimising. If you want to disable optimisations for clearer debugging you can use the "Switch Build Type" command in the Quick Access bar to switch to a Debug build, which will have fewer optimisations.

will-v-pi, 2025-03-12:
Closing due to no response, as I’m unable to reproduce the issue

closed 2025-03-12 by will-v-pi