raspberrypi/pico-vscode #167: No way to configure?

RudolphRiedel, 2025-02-23:
I just installed the extension and started to play with an example project.

What I can not figure out is, how do I configure the build?
I see the optimization is set to -O3, I want to use -O2.
How do I add -Wall -Wextra -pedantic?
How do I change to -std=c99?
And why is -mfloat-abi=softfp set when the M33 cores in the RP2350 do have a FPU?

The build.ninja file I am finding the flags in has "# CMAKE generated file: DO NOT EDIT!" - so what I am supposed to edit?
And even more importantly, why are there no options configuring this like this when setting up a new project?

Then uploading the code to the board, are we supposed to manually put the .uf2 files on the board?

will-v-pi, 2025-02-24:
You can set those options the same way you would for any CMake project, by setting them in the CMakeLists.txt file. You can search the internet for the variables for those, or ask on the forums if you're still having issues.

Regarding `-mfloat-abi=softfp`, you should ask that question on the main pico-sdk repository, as that's not related to this extension.

To upload you can either copy the UF2 files to the board in bootsel mode, click the Run button at the bottom while in bootsel mode to upload using picotool, or attach a debugger and upload it over SWD by clicking Debug in the Quick Action bar.

Closing this as none of these issues are related to the extension, they're just general questions about CMake and the Pico.

closed 2025-02-24 by will-v-pi

RudolphRiedel, 2025-02-24:
CMakeLists.txt:
# Generated Cmake Pico project file

This indicates that this file is not be messed with.
Ok, now I know to lower my expectations for this "Extension".

At least uploading without debugger works just fine using the "Run" button when the board is in bootsel mode, regardless of that there is no option anywhere that would indicate this.