raspberrypi/pico-vscode #169: vscode not listed prerequisites

anishp55, 2025-02-24:
**Describe the bug**
when installing the vscode extension, at least 2 other extensions are required for the pico extension to function.

**Extension Logs**
none, when code restarted with the plugin i was prompted to install them

**To Reproduce**
install a fresh vscode and make sure the old configs are not saved.
install pico-vscode, restart and be prompted to install the required extensions.

**Expected behavior**
while this is acceptable, they should really be called out as dependencies. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - OS: Win11
 - Architecture: x86_64

will-v-pi, 2025-02-24:
They are all listed as dependencies, you can see that in the `package.json`. I'm not sure why the dependencies aren't being installed on your system when you install the extension?

will-v-pi, 2025-03-04:
Closing this, as the prerequisites are listed as dependencies in `package.json` and are installed as dependencies when you install the extension.

closed 2025-03-04 by will-v-pi