raspberrypi/pico-vscode #170: unable to set sdk install location

anishp55, 2025-02-24:
i need to install the SDK for this to a different drive than where VScode is located.  Currently there is no option to do this, it appears there was a windows installer that may have let us do this in the past, but it is no longer maintained.

will-v-pi, 2025-02-24:
Sorry, this is not an option for this extension - the SDK and tools etc are all installed to `~/.pico-sdk`. We do not plan to add this because it would break the cross-system compatibility of projects, as the new path would have to be hard coded into the files.

What is the reason you can't have the SDK on the same drive as your home directory?

anishp55, 2025-02-25:
i need to do this for space reason and separation reason.  i tried to use a symlink but that didn't work.  also how would installing the toolchain to a different location break compatibility?

will-v-pi, 2025-02-25:
There are paths in the `.vscode/settings.json` and `.vscode/c_cpp_properties.json` files which don't support variable substitution with command variables (like is done in `.vscode/cmake-kits.json`), so those need to be set to a path relative to `userHome` in order for the project to work across different systems. I'll add the feature request label, but this isn't something we're planning to fix, as the symlink workaround should work fine.

I've just checked and a symlink does work for me - try the following command in an administrative PowerShell to create the symlink (replacing other_drive and other_folder with the place you want it to be stored):
```ps1
New-Item -ItemType SymbolicLink -Path ${env:USERPROFILE}\.pico-sdk -Target other_drive:\other_folder\.pico-sdk
```

Slion, 2025-03-19:
At my company we have sort of two home folders. One of them is a network drive. `USERPROFILE` is the usual local path but there is another one defined through various env vars as shown below:

```
HOMEDRIVE                   Z:
HOMEPATH                    \
HOMESHARE                   \\company\site\homedir\user
```

Everything was correctly installed in `USERPROFILE` but for some reason CMake Tools is looking for cmake on the network drive. Trying to create SymLink on the network drive fails as it is not supported.
To workaround the issue I had to copy cmake to the network drive.

anishp55, 2025-04-10:
> There are paths in the `.vscode/settings.json` and `.vscode/c_cpp_properties.json` files which don't support variable substitution with command variables (like is done in `.vscode/cmake-kits.json`), so those need to be set to a path relative to `userHome` in order for the project to work across different systems. I'll add the feature request label, but this isn't something we're planning to fix, as the symlink workaround should work fine.
> 
> I've just checked and a symlink does work for me - try the following command in an administrative PowerShell to create the symlink (replacing other_drive and other_folder with the place you want it to be stored):
> 
> New-Item -ItemType SymbolicLink -Path ${env:USERPROFILE}\.pico-sdk -Target other_drive:\other_folder\.pico-sdk

since i don't know anything about the internals of a vscode plugin i cannot fully comment on what your saying, but i do know that the esp-idf plugin supports letting me use an existing toolchain, or they will download one and install it.  with that being said i don't see why this plugin cannot follow a similar pattern.

will-v-pi, 2025-04-15:
> with that being said i don't see why this plugin cannot follow a similar pattern

This extension integrates with lots of other extensions, so needs features like variable substitution with command variables to be supported by the other extensions to use them; whereas that one doesn't integrate with lots of other extension, so can use custom everything.