raspberrypi/pico-vscode #171: Endless Reactivating Terminals

HopWorks, 2025-02-25:
I used this extension because the windows install of the PICO-SDK was riddled with issues. And I started a new project, "Create Pico Project" and am stuck at Endless Reactivating Terminals.

will-v-pi, 2025-02-26:
Please fill out the bug report template, and include extension logs and screenshots/video of your issue https://github.com/raspberrypi/pico-vscode/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

p-kager, 2025-02-26:
I had the same issue on Rocky Linux and it seems to be related to Python extension. There is a preview setting that has been set as default (I would not know since what version). 

I changed Settings -> Python:Locator to **js** instead of **native** it the problem went away.

[image]

HopWorks, 2025-02-28:
I just came here to post what you did p-kager and you beat me to it. Thank you sir for confirming this. I did in fact make this change and it did take care of the stuck 'reactivating terminals' and had no adverse effects on my build at all. 
I will also take the time to meet the criteria of posting an issue here, and I apologize for not following that process. I am always happy to help to get something fixed. And the process introduces a few things I did not know, so I can expand my knowledge of VSC and the pico extension also. THANKS!

closed 2025-02-28 by HopWorks