raspberrypi/pico-vscode #174: Build folder per board

Slion, 2025-03-08:
Is it possible to set a build folder per board and per configuration?

Like:
`build/picow/rel`
`build/picow/deb`
`build/pico2w/rel`
`build/pico2w/deb`

will-v-pi, 2025-03-11:
If you’re using the CMake Tools Extension integration, which uses that `cmake.buildDirectory` setting, then you won’t be using the Ninja task in `tasks.json`, so you should be fine just modifying the setting in `settings.json` and enabling the CMake Tools Extension integration.

I don’t think we’d plan to add this, but I’ll leave it open as a suggestion - for now, you can do this yourself by modifying that setting.

Slion, 2025-03-12:
For some reason I'm still using that Ninja task as I usually just do Ctrl+Shift+B to build and flash in one go:
Here is what my custom `tasks.json` looks like: 
```json
{
    "version": "2.0.0",
    "tasks": [
        {
            "label": "Compile Project",
            "type": "process",
            "isBuildCommand": true,
            "command": "${userHome}/.pico-sdk/ninja/v1.12.1/ninja",
            "args": ["-C", "${workspaceFolder}/build"],
            "group": "build",
            "presentation": {
                "reveal": "always",
                "panel": "dedicated"
            },
            "problemMatcher": "$gcc",
            "windows": {
                "command": "${env:USERPROFILE}/.pico-sdk/ninja/v1.12.1/ninja.exe"
            }
        },
        {
            "label": "Run Project",
            "type": "process",
            "command": "${env:HOME}/.pico-sdk/picotool/2.1.1/picotool/picotool",
            "args": [
                "load",
                "${command:raspberry-pi-pico.launchTargetPath}",
                "-fx"
            ],
            "presentation": {
                "reveal": "always",
                "panel": "dedicated"
            },
            "problemMatcher": [],
            "windows": {
                "command": "${env:USERPROFILE}/.pico-sdk/picotool/2.1.1/picotool/picotool.exe"
            }
        },
        {
            "label": "Flash",
            "type": "process",
            "command": "${userHome}/.pico-sdk/openocd/0.12.0+dev/openocd.exe",
            "args": [
                "-s",
                "${userHome}/.pico-sdk/openocd/0.12.0+dev/scripts",
                "-f",
                "interface/cmsis-dap.cfg",
                "-f",
                "target/${command:raspberry-pi-pico.getTarget}.cfg",
                "-c",
                "adapter speed 5000; program \"${command:raspberry-pi-pico.launchTargetPath}\" verify reset exit"
            ],
            "problemMatcher": [],
            "windows": {
                "command": "${env:USERPROFILE}/.pico-sdk/openocd/0.12.0+dev/openocd.exe",
            }
        }
    ]
}
```

Could I somehow replace it to use CMake directly?

Slion, 2025-03-15:
Thanks for the tip I did not know it could do that. I fixed the content of my `tasks.json` above because I initially posted the wrong one. I did a fresh import of my project, making sure I tick that Enable CMake Tools Extension integration option. Then to bind the flash task to Ctrl + Shift + B I added the following: 
```json
            "group": {
              "kind": "build",
              "isDefault": true
            }
```

Now I don't need to depend on and call the Compile Project Ninja task like I was doing above.

[image]

Slion, 2025-03-15:
Any reason why  `Enable CMake Tools Extension integration` is not on by default?

will-v-pi, 2025-03-15:
Yes - because we have no way to control the kit selection dialogue, so for a new user it’s simpler to use the tasks that this extension provides, as if you click the wrong kit or miss the dialogue then it gets quite confusing quite quickly.

Although I think we will move it out of advanced options, and make it remember the setting from the last time you created a project, as it is definitely very useful for more experienced developers.

will-v-pi, 2026-11-05:
Closing this now CMake Tools Extension integration is more supported and tested, and that is the way to go for things like custom build folders

closed 2026-11-05 by will-v-pi