raspberrypi/pico-vscode #179: The header file cannot be opened when creating a project in linux

Dingtaiqi, 2025-03-18:
I used arch, and I used vscode to create projects that went well but still had problems that I couldn't fix, even though I had fixed some bugs

[image]

I also have a successful compilation project on windows, so I checked it out and found that "/build/compile_commands.json" is empty in linux and has a lot of characters in winodws

[image]
[image]
[image]

will-v-pi, 2025-03-18:
Looks like your picotool download failed on Linux - try reloading the window, and check if `~/.pico-sdk/picotool/2.1.1/picotool/` exists and contains the `picotoolConfig.cmake` and `picotoolTargets.cmake` files, along with the `picotool` binary. If only some of the files are there, delete the `~/.pico-sdk/picotool/2.1.1` directory and reload the window again.

Dingtaiqi, 2025-03-18:
[image]
There should be no missing files for this picotool, which I downloaded and unzipped from github

will-v-pi, 2025-03-18:
You've downloaded the wrong thing, looks like you've downloaded the picotool repository, not a pre-built picotool.

You shouldn't have to download anything though, the extension does all the downloading for you, so you should delete the folder and reload the window, and it should download the pre-built picotool to that location

Dingtaiqi, 2025-03-19:
This is indeed the problem, and it was resolved after re-downloading. I am in China where the Internet is restricted and vscode has difficulty accessing github, so I tried to add the tool manually

will-v-pi, 2025-03-19:
See the other issue relating to use in China for help

closed 2025-03-19 by will-v-pi