raspberrypi/pico-vscode #181: The tinyusb SDK cannot be used properly

Dingtaiqi, 2025-04-01:
I tried to learn the tinyusb routine to add tinyusb to rp2040, but kept getting errors like "undefined reference to 'tud_hid_set_report_cb'" that would not compile

Tinyusb routines: https://github.com/hathach/tinyusb/blob/master/examples/device/hid_composite/src/tusb_config.h
[image]

Dingtaiqi, 2025-04-01:
[2/2] Linking CXX executable 114514.elf
FAILED: 114514.elf
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\Users\yinsh\.pico-sdk\toolchain\14_2_Rel1\bin\arm-none-eabi-g++.exe -mcpu=cortex-m0plus -mthumb -g -O3 -DNDEBUG -Wl,--build-id=none -Wl,-Map=114514.elf.map --specs=nosys.specs -Wl,--wrap=__clzsi2 -Wl,--wrap=__clzdi2 -Wl,--wrap=__ctzsi2 -Wl,--wrap=__popcountsi2 -Wl,--wrap=__popcountdi2 -Wl,--wrap=__clz -Wl,--wrap=__clzl -Wl,--wrap=__clzll -Wl,--wrap=__ctzdi2 -Wl,--wrap=__aeabi_idiv -Wl,--wrap=__aeabi_idivmod -Wl,--wrap=__aeabi_ldivmod -Wl,--wrap=__aeabi_uidiv -Wl,--wrap=__aeabi_uidivmod -Wl,--wrap=__aeabi_uldivmod -Wl,--wrap=__aeabi_dadd -Wl,--wrap=__aeabi_ddiv -Wl,--wrap=__aeabi_dmul -Wl,--wrap=__aeabi_drsub -Wl,--wrap=__aeabi_dsub -Wl,--wrap=__aeabi_cdcmpeq -Wl,--wrap=__aeabi_cdrcmple -Wl,--wrap=__aeabi_cdcmple -Wl,--wrap=__aeabi_dcmpeq -Wl,--wrap=__aeabi_dcmplt -Wl,--wrap=__aeabi_dcmple -Wl,--wrap=__aeabi_dcmpge -Wl,--wrap=__aeabi_dcmpgt -Wl,--wrap=__aeabi_dcmpun -Wl,--wrap=__aeabi_i2d -Wl,--wrap=__aeabi_l2d -Wl,--wrap=__aeabi_ui2d -Wl,--wrap=__aeabi_ul2d -Wl,--wrap=__aeabi_d2iz -Wl,--wrap=__aeabi_d2lz -Wl,--wrap=__aeabi_d2uiz -Wl,--wrap=__aeabi_d2ulz -Wl,--wrap=__aeabi_d2f -Wl,--wrap=sqrt -Wl,--wrap=cos -Wl,--wrap=sin -Wl,--wrap=tan -Wl,--wrap=atan2 -Wl,--wrap=exp -Wl,--wrap=log -Wl,--wrap=ldexp -Wl,--wrap=copysign -Wl,--wrap=trunc -Wl,--wrap=floor -Wl,--wrap=ceil -Wl,--wrap=round -Wl,--wrap=sincos -Wl,--wrap=asin -Wl,--wrap=acos -Wl,--wrap=atan -Wl,--wrap=sinh -Wl,--wrap=cosh -Wl,--wrap=tanh -Wl,--wrap=asinh -Wl,--wrap=acosh -Wl,--wrap=atanh -Wl,--wrap=exp2 -Wl,--wrap=log2 -Wl,--wrap=exp10 -Wl,--wrap=log10 -Wl,--wrap=pow -Wl,--wrap=powint -Wl,--wrap=hypot -Wl,--wrap=cbrt -Wl,--wrap=fmod -Wl,--wrap=drem -Wl,--wrap=remainder -Wl,--wrap=remquo -Wl,--wrap=expm1 -Wl,--wrap=log1p -Wl,--wrap=fma -Wl,--wrap=__aeabi_lmul -Wl,--wrap=__aeabi_fadd -Wl,--wrap=__aeabi_fdiv -Wl,--wrap=__aeabi_fmul -Wl,--wrap=__aeabi_frsub -Wl,--wrap=__aeabi_fsub -Wl,--wrap=__aeabi_cfcmpeq -Wl,--wrap=__aeabi_cfrcmple -Wl,--wrap=__aeabi_cfcmple -Wl,--wrap=__aeabi_fcmpeq -Wl,--wrap=__aeabi_fcmplt -Wl,--wrap=__aeabi_fcmple -Wl,--wrap=__aeabi_fcmpge -Wl,--wrap=__aeabi_fcmpgt -Wl,--wrap=__aeabi_fcmpun -Wl,--wrap=__aeabi_i2f -Wl,--wrap=__aeabi_ui2f -Wl,--wrap=__aeabi_f2iz -Wl,--wrap=__aeabi_f2uiz -Wl,--wrap=__aeabi_l2f -Wl,--wrap=__aeabi_ul2f -Wl,--wrap=__aeabi_f2lz -Wl,--wrap=__aeabi_f2ulz -Wl,--wrap=__aeabi_f2d -Wl,--wrap=sqrtf -Wl,--wrap=cosf -Wl,--wrap=sinf -Wl,--wrap=tanf -Wl,--wrap=atan2f -Wl,--wrap=expf -Wl,--wrap=logf -Wl,--wrap=sincosf -Wl,--wrap=ldexpf -Wl,--wrap=copysignf -Wl,--wrap=truncf -Wl,--wrap=floorf -Wl,--wrap=ceilf -Wl,--wrap=roundf -Wl,--wrap=asinf -Wl,--wrap=acosf -Wl,--wrap=atanf -Wl,--wrap=sinhf -Wl,--wrap=coshf -Wl,--wrap=tanhf -Wl,--wrap=asinhf -Wl,--wrap=acoshf -Wl,--wrap=atanhf -Wl,--wrap=exp2f -Wl,--wrap=log2f -Wl,--wrap=exp10f -Wl,--wrap=log10f -Wl,--wrap=powf -Wl,--wrap=powintf -Wl,--wrap=hypotf -Wl,--wrap=cbrtf -Wl,--wrap=fmodf -Wl,--wrap=dremf -Wl,--wrap=remainderf -Wl,--wrap=remquof -Wl,--wrap=expm1f -Wl,--wrap=log1pf -Wl,--wrap=fmaf -Wl,--wrap=malloc -Wl,--wrap=calloc -Wl,--wrap=realloc -Wl,--wrap=free -Wl,--wrap=memcpy -Wl,--wrap=memset -Wl,--wrap=__aeabi_memcpy -Wl,--wrap=__aeabi_memset -Wl,--wrap=__aeabi_memcpy4 -Wl,--wrap=__aeabi_memset4 -Wl,--wrap=__aeabi_memcpy8 -Wl,--wrap=__aeabi_memset8 -Wl,-LC:/Users/yinsh/114514/build -Wl,--script=C:/Users/yinsh/.pico-sdk/sdk/2.1.1/src/rp2_common/pico_crt0/rp2040/memmap_default.ld -Wl,-z,max-page-size=4096 -Wl,--gc-sections -Wl,--no-warn-rwx-segments -Wl,--wrap=sprintf -Wl,--wrap=snprintf -Wl,--wrap=vsnprintf -Wl,--wrap=printf -Wl,--wrap=vprintf -Wl,--wrap=puts -Wl,--wrap=putchar -Wl,--wrap=getchar @CMakeFiles\114514.rsp -o 114514.elf && C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\yinsh\114514\build && C:\Users\yinsh\.pico-sdk\toolchain\14_2_Rel1\bin\arm-none-eabi-objdump.exe -h C:/Users/yinsh/114514/build/114514.elf > 114514.dis && C:\Users\yinsh\.pico-sdk\toolchain\14_2_Rel1\bin\arm-none-eabi-objdump.exe -d C:/Users/yinsh/114514/build/114514.elf >> 114514.dis && cd /D C:\Users\yinsh\114514\build && C:\Users\yinsh\.pico-sdk\toolchain\14_2_Rel1\bin\arm-none-eabi-objcopy.exe -Oihex C:/Users/yinsh/114514/build/114514.elf 114514.hex && cd /D C:\Users\yinsh\114514\build && C:\Users\yinsh\.pico-sdk\toolchain\14_2_Rel1\bin\arm-none-eabi-objcopy.exe -Obinary C:/Users/yinsh/114514/build/114514.elf 114514.bin && cd /D C:\Users\yinsh\114514\build && C:\Users\yinsh\.pico-sdk\picotool\2.1.1\picotool\picotool.exe uf2 convert --quiet C:/Users/yinsh/114514/build/114514.elf 114514.uf2 --family rp2040""
C:/Users/yinsh/.pico-sdk/toolchain/14_2_Rel1/bin/../lib/gcc/arm-none-eabi/14.2.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/114514.dir/C_/Users/yinsh/.pico-sdk/sdk/2.1.1/lib/tinyusb/src/class/hid/hid_device.c.obj: in function `hidd_control_xfer_cb':
C:/Users/yinsh/.pico-sdk/sdk/2.1.1/lib/tinyusb/src/class/hid/hid_device.c:284:(.text.hidd_control_xfer_cb+0x58): undefined reference to `tud_hid_descriptor_report_cb'
C:/Users/yinsh/.pico-sdk/toolchain/14_2_Rel1/bin/../lib/gcc/arm-none-eabi/14.2.1/../../../../arm-none-eabi/bin/ld.exe: C:/Users/yinsh/.pico-sdk/sdk/2.1.1/lib/tinyusb/src/class/hid/hid_device.c:333:(.text.hidd_control_xfer_cb+0xd2): undefined reference to `tud_hid_set_report_cb'
[... cut ...] t_report_cb'
C:/Users/yinsh/.pico-sdk/toolchain/14_2_Rel1/bin/../lib/gcc/arm-none-eabi/14.2.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/114514.dir/C_/Users/yinsh/.pico-sdk/sdk/2.1.1/lib/tinyusb/src/class/hid/hid_device.c.obj: in function `hidd_xfer_cb':
C:/Users/yinsh/.pico-sdk/sdk/2.1.1/lib/tinyusb/src/class/hid/hid_device.c:401:(.text.hidd_xfer_cb+0x30): undefined reference to `tud_hid_set_report_cb'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

will-v-pi, 2025-04-01:
This is not an issue with the VS Code extension, so closing it. The best place to ask about it would probably be on our forums https://forums.raspberrypi.com/

That being said, looking at that example, those functions are defined in the main.c file, so if you haven’t copied those definitions over then that would explain your issue

closed 2025-04-01 by will-v-pi

Dingtaiqi, 2025-04-01:
There's still a problem. The SDK of tinyusb in 2.1.1SDK is different from the official sdk part, resulting in the direct use of 2.1.1 tinyusb can not successfully use tinyusb example
https://github.com/hathach/tinyusb/blob/master/src/class/hid/hid.h#L340

[image]