raspberrypi/pico-vscode #184: Where is the "RUN" command

Dingtaiqi, 2025-04-08:
I in compiling "https://github.com/hathach/tinyusb/tree/86ad6e56c1700e85f1c5678607a762cfe3aa2f47/examples/device/hid_multiple_inter face
Although it can compile, it cannot be debugged through the "RUN" at the bottom

[image]

I looked for the order as best I could but I couldn't find it

will-v-pi, 2025-04-08:
Your error is that your project name is defined as a variable, which is not supported unless you enable the [CMake Tools Extension Integration](https://github.com/raspberrypi/pico-vscode?tab=readme-ov-file#cmake-tools-extension-integration) - you can see this because it’s looking for `${PROJECT}.elf`

To answer you question in the title, please see the [Getting Started Guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf#page9) for run and debug instructions

closed 2025-04-08 by will-v-pi

zzn3y2, 2025-04-09:
Use the RUN AND DEBUG (Ctrl + Shift + D) to connect the debugger and then the Continue (F5) to run the program.

MarcoMagana, 2025-04-09:
Use the RUN AND DEBUG (Ctrl + Shift + D) to connect the debugger and then the Continue (F5) to run the program.