raspberrypi/pico-vscode #187: MicroPython - I can't open a .py ready program

Gustavomurta, 2025-05-02:
Raspberry Pico Project extension works fine with C++ projects. But I didn't find how to open a ready-made MicroPython program. I only found the New MicroPython project option (blink.py). 
Import projects - only for C++ projects. 
New project from Sample - only for C++ projects.
Thanks.

paulober, 2025-05-02:
For MicroPython support there is the MicroPico extension which should have been installed with pico-vscode. You can activate it for your MicroPython project by opening the folder and running the "Setup MicroPico Project" command either in the explorer context menu or the command palette.

paulober, 2025-05-02:
> So I understand that the option to open a ready-made project in the **Raspberry Pico Project** extension is not yet supported.

Right, but the create MicroPython project option is also provided by MicroPico so pico-vscode is only for C/C++ at the moment.

closed 2025-05-02 by paulober

Gustavomurta, 2025-05-03:
I found detailed instructions in this manual:
[Raspberry Pi Pico-series Python SDK](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-python-sdk.pdf)

**Chapter 4.2. Using Visual Studio Code**