raspberrypi/pico-vscode #196: release 0.17.7 (today's) issue with pico2_w

jancumps, 2025-07-25:
**Describe the bug**
I'm using the extension to wxork on Pico and Pico2 boards. After activating latest version, it includes boards/pico2.h instead of pico2_w.h.

I deleted build. Cleaned CMAke, Confugured CMake. Compiled.

**Extension Logs**
Compilation works. But it uses pico2.h
[image]

**To Reproduce**
1. Have a project that works with the extension, has pico2_w selected, and uses the usual blinky construct:
// Initialise led
static void pico_init_led(void) {
#if defined PICO_DEFAULT_LED_PIN
    gpio_init(PICO_DEFAULT_LED_PIN);
    gpio_set_dir(PICO_DEFAULT_LED_PIN, GPIO_OUT);
#elif defined(CYW43_WL_GPIO_LED_PIN)
    hard_assert(cyw43_arch_init() == PICO_OK);
    pico_set_led(false); // make sure cyw43 is started
#endif
}
2. delete build folder to avoid remnants of previous compilations
3. clean cmake with extension
4. double check that pico2_w is selected. Correct if not
5. use extension to configure cmake
6. compile
7. at this point, VSCode and toolchain should use pico2_w, but don't. boards/pico2.h is included.

**Screenshots**
[image]

**Platform (please complete the following information):**
 - OS: Windows11, latest VSCode, today's extension
 - Architecture: x86

**Additional context**
I used this project yesterday without issues. pico2_w header was used (e.g. for the on-board LED functionality) Today VSCode asked to refresh the Pico extension. That's when this changed.

will-v-pi, 2025-07-25:
Just to confirm, did you reload the window when prompted after switching the board? Alternatively close & reopen the window instead of reloading.

Is your binary being compiled for the correct board and this is just an intellisense problem, or is the binary being compiled for the wrong board? You can check this by running `picotool info -l build/blink.uf2` (replace blink with your program name).

Also, you should switch steps 3 & 4 so the clean occurs with the correct board, and can skip step 5 because clean CMake performs a configure after cleaning

The changes from 0.17.6 to 0.17.7 mostly relate to new projects rather than existing projects, so it’s unlikely that has caused this issue

jancumps, 2025-07-25:
- I closed all windows. Restarted VSCode
- The binary is for pico2

[image]
- I will try and report back
- reporting what I see at the moment. Will report back after I tried question 3

jancumps, 2025-07-25:
After switching step 3 & 4:  (I had to set back to Debug after Clean - because it switches the extension to Release
extension settings:  

[image]

compilation successful:  

[image]

still pico2. Not pico2_w

[image]

jancumps, 2025-07-26:
I've attached my project (without the build dir for size)

CMakeLists.txt:

```
# Generated Cmake Pico project file

cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)

# == DO NOT EDIT THE FOLLOWING LINES for the Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.1.1)
set(toolchainVersion RISCV_RPI_2_0_0_5)
set(picotoolVersion 2.1.1)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================
# set(PICO_BOARD pico2_w CACHE STRING "Board type") # set by extension

include(FetchContent)
# FetchContent_Declare(callbackmanager
#   GIT_REPOSITORY "https://github.com/jancumps/callbackmanager.git"
#   GIT_TAG "origin/main"
#   SOURCE_SUBDIR =
# )
# FetchContent_MakeAvailable(callbackmanager)

FetchContent_Declare(freertos
  GIT_REPOSITORY "https://github.com/FreeRTOS/FreeRTOS-Kernel.git"
  GIT_TAG "origin/main"
  SOURCE_SUBDIR =
)
FetchContent_MakeAvailable(freertos)
set(FREERTOS_KERNEL_PATH ${freertos_SOURCE_DIR})

# Pull in Raspberry Pi Pico SDK (must be before project)
include(pico_sdk_import.cmake)

# Pull in FreeRTOS
include(FreeRTOS_Kernel_import.cmake)

project(pico2_risk_freertos C CXX ASM)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

add_executable(${CMAKE_PROJECT_NAME})

# target_compile_definitions(${CMAKE_PROJECT_NAME} PRIVATE PICO_DEFAULT_LED_PIN=22) # pico eurocard

target_sources(${CMAKE_PROJECT_NAME}
        PUBLIC
        ${CMAKE_CURRENT_SOURCE_DIR}/source/main.c
        ${CMAKE_CURRENT_SOURCE_DIR}/source/FreeRTOSHooks.c
)

target_include_directories(${CMAKE_PROJECT_NAME} PRIVATE
      ${CMAKE_CURRENT_SOURCE_DIR}
      ${CMAKE_CURRENT_SOURCE_DIR}/source
)

target_link_libraries(${CMAKE_PROJECT_NAME} PRIVATE
        pico_stdlib
        FreeRTOS-Kernel FreeRTOS-Kernel-Heap4 
        hardware_gpio
        pico_async_context_freertos
)
if(PICO_CYW43_SUPPORTED)
    # For led support on pico_w
    target_link_libraries(${CMAKE_PROJECT_NAME} PRIVATE
        pico_cyw43_arch_none
    )
endif()

pico_add_extra_outputs(pico2_risk_freertos)

```
For downgrading the extension, I'll have to read up on how to do that ...

jancumps, 2025-07-27:
Solved. It's my own problem, in the makefile:

# set(PICO_BOARD pico2_w CACHE STRING "Board type") # set by extension
Removed the comment, and the switching the board works:

[image]

closed 2025-07-27 by jancumps