raspberrypi/pico-vscode #203: Outdated openocd executable installed with pico-sdk v2.2.0

ohir, 2025-08-26:
With toolchain installed by the pico-vscode extension the openocd server `~/.pico-sdk/openocd/0.12.0+dev/openocd` is not the one tagged for the `2.2.0` pico-sdk in this [commit](https://github.com/raspberrypi/openocd/commit/30c3d077f281876286a7aa37afbd411d4bd1667e) from Jul 16 2025.  This implies that openocd in the bundle, as tagged, should use `libjim v0.83`. The older version that got installed is compiled against `libjim v0.79` that is not avaliable on newer Linux distros.

Setup: fresh VSCode (codium) installation, just with set of extensions added by the pico-vscode.

#### Workaround for debian based systems:
```
wget http://ftp.de.debian.org/debian/pool/main/j/jimtcl/libjim0.79_0.79+dfsg0-2_amd64.deb
sudo dpkg -i libjim0.79_0.79+dfsg0-2_amd64.deb
```

will-v-pi, 2025-08-26:
Try running the Update OpenOCD command (Ctrl+Shift+P then search for Update OpenOCD) - it should update to the latest one in use by the extension. This should also be the default version installed when you first run the extension (provided you first install version 0.17.7 or higher).

Note that you may still experience this libjim issue when using that latest OpenOCD, as OpenOCD no longer uses it’s libjim submodule (which the commit you reference updated), and instead uses the system libjim, so may only work on Linux distros with the same version as the one we compiled it with (see the pico-sdk-tools repo for how the pre-compiled binaries are built).

jiserafini, 2025-09-03:
The same problem with Fedora 42 and Arch Linux.

will-v-pi, 2025-09-05:
> I should underscore that I opened issue right after fresh install of the extension taken from the github releases. OpenOCD was installed by the extension setup.

Thanks for that confirmation - I've renamed the issue to reflect that, as it is installing the correct OpenOCD, it's just that OpenOCD is compiled against an older libjim than your system has

will-v-pi, 2025-09-08:
Reopening until the extension is updated to use the fixed OpenOCD

art51go, 2025-09-09:
I just downgraded my libjim to version 0.79 as you recommended. When I am starting a debug session now, I get the same "GDB Server Quit Unexpectedly" with "Error: nm-multiarch failed! ..." the Terminal Message is now:

`Open On-Chip Debugger 0.12.0+dev (2025-07-17-12:43)

Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'gdb port', not 'gdb_port'
DEPRECATED! use 'tcl port' not 'tcl_port'
DEPRECATED! use 'telnet port', not 'telnet_port'
CDLiveWatchSetup
Info : [rp2040.core0] Hardware thread awareness created
Info : [rp2040.core1] Hardware thread awareness created
adapter speed: 5000 kHz
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Error: unable to find a matching CMSIS-DAP device`

will-v-pi, 2025-09-09:
> I just downgraded my libjim to version 0.79 as you recommended. When I am starting a debug session now, I get the same "GDB Server Quit Unexpectedly" with "Error: nm-multiarch failed! ..." the Terminal Message is now:

> Error: unable to find a matching CMSIS-DAP device`

This is an unrelated issue - OpenOCD can't see your debugprobe, likely either because it isn't plugged in, or you don't have udev rules installed to use it without `sudo` (see the Linux requirements in the readme)

art51go, 2025-09-09:
Yes. The udev rules were missing. Everything is working now. Thank You.

closed 2025-09-15 by will-v-pi